python-control/python-control #464: root_locus() plot figure size issue - ipynotebook

RicardoSQueen, 2020-11-03:
Hello,
I'm currently running python-control 0.8.3 on python 3.8.5.
When plotting 'root_locus' in a interactive python notebook, I get a plot way offset to the right, according to the image below:
[image]
The unexpected for me, though, is that when the exact same code directly from the terminal, the plot was perfect.
The plot issues appears on every interactive notebook environment: jupyter notebook, jupyter lab and visual studio code's jupyter implementation. I'm running these versions of jupyter:
notebook                       6.1.4
jupyter_client                 6.1.7
jupyter_contrib_core     0.3.3
jupyter_core                  4.6.3
jupyter_latex_envs        1.4.6
jupyterlab                      2.2.8
jupyterlab_pygments     0.1.2
jupyterlab_server          1.2.0

here is the code for reproducing:
```
from control import *
import matplotlib.pyplot as plt

num = [1, 32, 247]
den = [1, 37,385, 118, 0]
x1 = tf(num, den)
rlist, klist = root_locus(x1)
plt.show()
```

sawyerbfuller, 2020-11-04:
Hi Ricardo, thanks for pointing out the issue. I think this has been fixed in #413 and is now in the dev branch (see screenshot below running the dev branch). Release 0.8.4 is right around the corner and includes these fixes. 

[image]

RicardoSQueen, 2020-11-15:
Hey Sawyer,
Sorry, I tried finding an issue related to my issue but I didn't find any.. oh well. Awesome to know that!
I did install the dev branch over here and it seems that it's still not ok. Here are some screenshots:
[image]
[image]
[image]
The plot is all white to me.. And it took a little while (longer than usual)
Running your code worked out great:
[image]
And I tried running the same transfer function (L2) but continuous and I got an error:
[image]
 So there seems to be an issue in continuous transfer functions.. Below is the error message I got:
`ValueError: Image size of 1932716x278 pixels is too large. It must be less than 2^16 in each direction.
`
And the full traceback is attached.
[traceback.txt](https://github.com/python-control/python-control/files/5541608/traceback.txt)

murrayrm, 2020-12-26:
Confirming that I get results similar to @RicardoSQueen: when running the code at the top for reproducing the error, I am able to get a plot in ipython but not in a Jupyter notebook.  The example that @sawyerbfuller gives works in both cases.  This seems to be triggered by something we are doing in `rlocus`, but not clear that the fix is anything obvious.

I don't think this this is something we can fix prior to release of 0.8.4, so pushing out to 0.9.x for now.

murrayrm, 2021-01-09:
I chased down the problem (with lots of annoying sorting out what uncommented code was doing).  The problem boiled down to labels for the constant natural frequency lines being annotated outside of the bounds of the plot.  In python/ipython this seems to be handled OK, but in Jupyter notebooks it messes up the aspect ratio of the resulting plot.  Final fix was just to check the location version the limits and supress the annotation.  PR coming.

closed 2021-01-14 by murrayrm