python-control/python-control #465: control.margin() gives wrong result

AchmadFathoni, 2020-11-05:
```
import control
import matplotlib.pyplot as plt

Gp = control.tf([2],[1,3,2,0])
Gz = control.c2d(Gp,0.05)

mag, phase, omega = control.bode(Gz)

print(control.margin(Gz))
print(control.margin([mag, phase, omega]))

plt.grid()
plt.show()

```
The example above prints 
```
(31.505480557979414, inf, 3.7736323313549196, nan)
(inf, 177.40890111291884, nan, 0.749329697586152)
```
According to bode plot the result should be
```
(inf,31.7,nan,0.75)
```

AchmadFathoni, 2020-11-05:
Matlab result:
`(2.79, 31.5, 1.36, 0.749)`

AchmadFathoni, 2020-11-05:
This is another way to get the margins
```
import control as ct 
import numpy as np
import matplotlib.pyplot as plt
import math
from scipy import interpolate

Gp = ct.tf([2],[1,3,2,0])
Gz = ct.c2d(Gp, 0.05)

mag, phase, omega = ct.bode(Gz)

p_w = interpolate.interp1d(phase, omega)
p_m = interpolate.interp1d(phase, mag)
print("Gain margin %.2f at %.2f rad/s"%(1/p_m(-np.pi), p_w(-np.pi)))

m_w = interpolate.interp1d(mag, omega)
m_p = interpolate.interp1d(mag, phase)
print("Phase margin %.3f\N{DEGREE SIGN} at %.3f rad/s"%(math.degrees(m_p(1)+np.pi), m_w(1)))

```

gives:
```
Gain margin 2.77 at 1.37 rad/s
Phase margin 31.394° at 0.753 rad/s
```

bnavigator, 2020-11-06:
I can confirm the same values with the current master branch. Note that there is an inconsistency between `bode()`/`bode_plot()` which returns the phase in radians and `margin()`/`stability_margins()`, which expects it in degrees:

```
In [11]: control.margin((mag, phase/np.pi*180, omega))
Out[11]: (2.792304251278397, 31.54075892910612, 1.36384900433416, 0.7493353337779878)
```

Get the same values with `bode_plot(..., margins=True)`.

The result for the discrete transfer function looks wrong though. Looks like the evaluation does not take into account the discretized time in https://github.com/python-control/python-control/blob/2d7aad0d98eda40f416a42ee96b638a6b409bc99/control/margins.py#L154-L210

The unit tests don't test any discrete transfer function either.

PR #469 (2020-12-21): Stability margins for discrete time systems
My attempt to fix #465 

If someone has an idea how to solve `min |1 + num(z) / den(z) |`  with `z=exp(i w dt)` analytically,  that would be great. I opted for a numerical optimization instead.
closed 2020-12-21 by bnavigator