python-control/python-control #466: (Win 32-bit ) ImportError: cannot import name 'frange' from 'matplotlib.mlab'

PizzocaroSolomon, 2020-11-07:
Hi everyone! :) 
I have the same issue as #339 . I understand that the problem is my old control package version (0.7.0) that calls an object that in the new version of matplotlib was removed. 
I have installed the package with conda. running the command : 
`conda install -c conda-forge control`
Since I have Anaconda for Windows 32-bit (it's not a mistake ... I actually have W 32 bit :-1: ) it automatically install the version 0.7.0 of the control package. 

I read in the README that the package is available for Windows 64-bit. This means that I can't install the updated version ( > 0.8.2) on windows 32-bit ?

thanks for your help! 
I hope the best for you all guys! :) 

Solomon

bnavigator, 2020-11-07:
Conda-forge dropped win32 a long time ago https://github.com/conda-forge/staged-recipes/issues/5640. That's probably why you only get really old packages from there.

python-control itself is a pure python library. You should be able to use it on 32-bit, given you have all the dependencies. Installing control directly from PyPI (pip) or using the archive from the Github releases page should do.

PizzocaroSolomon, 2020-11-19:
> Conda-forge dropped win32 a long time ago [conda-forge/staged-recipes#5640](https://github.com/conda-forge/staged-recipes/issues/5640). That's probably why you only get really old packages from there.
> 
> python-control itself is a pure python library. You should be able to use it on 32-bit, given you have all the dependencies. Installing control directly from PyPI (pip) or using the archive from the Github releases page should do.

Thank you! This worked!! :).

closed 2020-11-19 by PizzocaroSolomon