python-control/python-control #467: Phase shift in bode plot

SebastienDeriaz, 2020-11-10:
Given the transfer function
G = 1/s^3
We should see a constant negative phase of -270°, which is confirmed in matlab

[image]

However, both **control.bode** and **control.matlab.bode** give 90° instead of -270°. Although technically the same value, it does get confusing.

```
import control
import control.matlab
s = control.tf('s')

G = 1/s**3

control.bode(G);
```
[image]

```
control.matlab.bode(G);
```
[image]

What i've tried so far
- [x] using different transfer functions (1/s^4 or 1/s^2 * 1/(s+10))
- [x] specifying "omega" with a frequency range (as suggested here : https://ch.mathworks.com/matlabcentral/answers/99531-why-is-the-phase-that-is-returned-by-the-bode-function-wrapping-in-the-control-system-toolbox-5-1-r )
- [x] extending the "omega" range

Is it possible to get the "absolute" phase of the transfer function ? If not, i would suggest that `control.matlab.bode` be corrected as to reflect the behavior of matlab's bode

Thanks for your help !

SebastienDeriaz, 2020-11-10:
My (limited) understanding of the function is that **bode** uses **freqresp** which uses **evalfr** and then **numpy.angle**. Since **evalfr** returns a complex number it isn't directly possible to get the "absolute" angle so the issue i'm reporting seems to require a dedicated code to determine a "reference phase" of some sort.
My excuses for my current level of python (or gitHub) doesn't lead me much further than that

bnavigator, 2020-11-10:
Thanks, for bringing this up.

Related issue: #388 

> My excuses for my current level of python (or gitHub) doesn't lead me much further than that

You are nevertheless very welcome to practice your Python and GitHub skills by trying to fix the issue and submitting a pull request :student:

ilayn, 2020-11-10:
The phase correction is only possible when you have the pole-zero structure and the behavior `w->0`, so you have to compute the initial phase to get the right amount of shift which is typically a minus sign attached to the `atan2` computation. 

When you supply a custom frequency grid then it is trickier to snap to the correct phase since then you have to evaluate more stuff to get to that grid from the pole-zero structure (if there is any, say FRD objects don't have it).

bnavigator, 2021-01-03:
Resolved in #494. If you think this is not solved, please reopen.

closed 2021-01-03 by bnavigator

Juan-Gg, 2023-08-09:
It seems to work with poles, but it does not with zeros.

For a single zero a constant phase of 90º would be expected, it gives -270º instead.

```
from matplotlib import pyplot as plt 
from control.matlab import *  # MATLAB-like functions

%matplotlib widget
s = tf('s')
G = (s)
plt.figure()
bode(G, wrap_phase = False)
plt.show()
```
[image]