python-control/python-control #470: system step response is incorrect for denominator with 0 at x^0

jmrubillon, 2020-11-17:
Take a simple function `f(t) = 1-0.2*exp(-t)`
The Laplace transform of which is `(0.8s+1)/(s^2+s)`

The following Notebook code **should** produce a stepped exponential starting at 0.8 (for t=0) and finishing at 1 (for t >= 6-ish).

```
%pylab inline
import control
sys = control.tf([0.8, 1], [1, 1, 0])
T, yout = control.step_response(sys)
plt.plot(T, yout)
```

However it draws a straight line `y = t`

bnavigator, 2020-11-17:
This is correct behavior. 

If you have a pole at 0, your system is only marginally stable.

In mechanical terms: Your natural frequency is 0 and you have rigid body motion. Excited by a constant force, the system runs away in a straight line, working against the damping of the second pole.

You get the plot, which you expect with `control.impulse_response(sys)`, though. The marginally stable system does not return to zero, but settles at the constant 1.

closed 2020-11-17 by bnavigator

ilayn, 2020-11-17:
You need to convolve  1-0.2*exp(-t) with an all ones function. What you expect is as mentioned is the convolution with a dirac delta function. In other words step plot is showing you `[(0.8s+1)/(s^2+s)]*(1/s)`