python-control/python-control #471: I can not get the correct output using control library

Klarom, 2020-12-09:
**Hi! This is a code that provides me from my course. Originally our instructor use it like this but i do not have mpcontrol and mpcontrolPc modules;**

[image]
 
**So i rearanged the code like this. But i get output completely wrong. In adittion i must say that i am new to python.**

[image]

bnavigator, 2020-12-09:
Hi @Klarom,

First, make sure that you use the correct parameters and expect the correct return values when replacing the mpcontrol functions by the python-control equivalents. 

You can check the [docs online](https://python-control.readthedocs.io/en/0.8.3/) or use the help function in Spyder. If you move the cursor to the function or method in question and press the help shortcut (CTRL+I by default, if I am not mistaken), the help pane will show the correct call signature.

control.bode is actually a redict to [.bode_plot](https://python-control.readthedocs.io/en/0.8.3/generated/control.bode_plot.html):
```
Definition : bode_plot(syslist, omega=None, plot=True, omega_limits=None, omega_num=None, margins=None, *args, **kwargs)`
...
omega (list) – List of frequencies in rad/sec to be used for frequency response
```

You request a single frequency of 1000. Adjust line 20 accordingly.

With spyder and python-control you are well equipped to tackle this course and learn both programming in Python as well as the basics of control systems. Have fun!

juanodecc, 2020-12-09:
you are plot the frequiency omega = 1000. try just [mag,phase, freq] = ctrl.bode(gcl)

Klarom, 2020-12-09:
Both comments helped me much thank you! I do not know that if i can ask another question here but i have one more question but for different code.

```
import control

#System
Ts = 0.0005
num1 = [0.00943, 0.01886, 0.00943]
den1 = [1, -1.8188, 0.8224]
#Transfer function of the system
g = control.tf(num1,den1,Ts)
print(g)
#Transfer function of the lag controller
num2 = [0.6514, -0.6514*0.9950]
den2 = [1, -1]
c = control.tf(num2,den2,Ts)
#Transfer function of the feedback system
num3 = [1]
den3 = [1]
h = control.tf(num3,den3,Ts)
#Transfer function of the closed-loop controlled system
gol = control.series(g,c)
gcl = control.feedback(gol,h)
#Step input signal
N = 500
x = control.step_signal(N, 1)
#Calculate the output signal
y = control.lsim(gcl, x)
```

**Here i get the error "module 'control' has no attribute 'step_signal' " i checked [this link](https://python-control.readthedocs.io/en/0.8.3/) but couldn't find a replacement for this line i tried some but no luck with it. `[x = control.step_signal(N, 1)]`**

bnavigator, 2020-12-09:
Try [step_response](https://python-control.readthedocs.io/en/0.8.3/generated/control.step_response.html)

```python
import numpy as np
T = np.arange(0, Ts*N, Ts)
Tout, y = control.step_response(gcl, T)
```

Klarom, 2020-12-09:
I tried and yes, I got output like that. But how can I plot the output in terms of N? Sorry i bombarded you with questions but the information i could find about this python-control is limited. With matlab I think it would look like this but i don't know how to do it on Python: 
[image]

bnavigator, 2020-12-09:
```python
from matplotlib import pyplot as plt

# x-Axis n
plt.plot(y)
plt.xlabel('n')
plt.ylabel('y[n]')

plt.figure()
plt.plot(Tout, y)
plt.xlabel('t [s]')
plt.ylabel('y(t)')
```

closed 2020-12-09 by bnavigator

Klarom, 2020-12-15:
I don't want to post another question here (or i should post a new one?) I have a quick question. Here i have a problem with plotting step response. My graph is look like this and i have no idea whats wrong here. 

```
import control
import numpy as np
import matplotlib.pyplot as plt
Ts = 0.5
G1 = control.tf([60], [1,0.1])
G2 = control.tf([0.5, 3], [1, 5, 12])
G3 = control.tf([1], [1, 5])
Gs= G1*G2*G3
Gz = control.c2d(Gs,Ts, method= 'zoh')

print(Gz)
print(Gs)

cltf=(Gs/(1+Gs))
Zcltf=(Gz/(1+Gz))

T = np.arange(0, 10)
za = control.step_response(cltf, T)

zb = control.step_response(Zcltf, T) 
plt.plot(zb)
plt.plot(za)
```

[image]