python-control/python-control #472: too many values to unpack (expected 2)

mistano, 2020-12-10:
Te = 1

G = tf(1,[1,1,0])
Gz = c2d(G, Te, 'zoh')
C1z = tf([1],[1,-1],Te)
C2z = (1/b1)*tf([1,-a],[1,-b0/b1],Te) 
C3z = tf([0.2488,-0.2134],[1],Te) 
Cz = C1z*C2z*C3z

sys1 = Cz*Gz/(1+Cz*Gz)
dts = 1
u1ds = np.zeros(tempsSimulation.shape);
u1ds[0] = 1/dts

ctrl.forced_response(sys1, tempsSimulation, u1ds, 0)

Hello, there is my code, and i'm getting the error : too many values to unpack (expected 2)
Can you help me please, also, i have to print with matplotlib the curves from a parabol input, that's why i am using forced_response, can you help me with it too please ?

Thanks a lot !

bnavigator, 2020-12-10:
Hello. The code is incomplete and not oconsistent with your error message.

 -  `a`, `b0`, `b1`, `tempsSimulation` are not defined. (Also `ctrl`, `np`, `tf`, `c2d`)
 - Please use [markdown code tags](https://guides.github.com/features/mastering-markdown/) so that the `*` does not get swallowed
- You most likeley want to do something with the response. Assign the return values of forced_response to `T, yout, xout`. These are 3 values in a tuple. "expected 2" indicates, that you actually want to assign to only 2 variables.  But that does not fit your posted code.

mistano, 2020-12-10:
`
import control as ctrl
import numpy as np
from control.matlab import *
import matplotlib.pyplot as plt

Te = 1
tau = 1

a = np.exp(-Te/tau)
b1 = Te - tau * (1 - a)
b0 = a*Te - tau * (1-a)

G = tf(1,[1,1,0])
Gz = c2d(G, Te, 'zoh')

C1z = tf([1],[1,-1],Te)
C2z = (1/b1)*tf([1,-a],[1,-b0/b1],Te)
C3z = tf([0.2488,-0.2134],[1],Te)

Cz = C1z*C2z*C3z

TempsDeReponseEstime = 30
tempsSimulation = np.arange(0,TempsDeReponseEstime,Te)

sys1 = Cz*Gz/(1+Cz*Gz)
dts = 1
u1ds = np.zeros(tempsSimulation.shape);
u1ds[0] = 1/dts

ctrl.forced_response(sys1, tempsSimulation, u1ds, 0)
`

There it is, hoping everything you need to help me is there :)

bnavigator, 2020-12-10:
I see no unpack error with this code. The last line returns a tuple for the following plot:

```python
T, yout, xout = _

plt.subplot(2,1,1)
plt.plot(T, yout)
plt.subplot(2,1,2)
plt.plot(T, xout.T)
```

[image]

Could you please post your backtrace?

mistano, 2020-12-10:
Hello, it seems to work when adding xout, i'm verry sorry about it, thanks a lot for your help !

closed 2020-12-10 by mistano