python-control/python-control #493: sumblk doesnt exists for connection in matlab compatibility module

k10blogger, 2020-12-31:
From upgrading from 0.7.0 to 0.8.0 of python control library. I cannot find the sumblk function in matlab compatibility module in the version 0.8.0

Here is a link to documentation of previous release.
https://python-control.readthedocs.io/en/0.7.0/matlab_strings.html#system-interconnections

I cannot seem to find this in new release. 
I am also not able see the reason for this exclusion.

sawyerbfuller, 2020-12-31:
Hi siddarth, that feature did not have an asterisk (*) next to it, indicating that it wasn’t yet implemented. 

I agree it would be a very nice feature to have and have I have used it myself, created as follows: 

sumblk = control.ss([[0,0.],[0,0]], [[0,0.],[0,0]], [[0,0.],[0,0]], [[1,  0],[1,  1]])

murrayrm, 2021-01-09:
A few comments on this:

* A less "stateful" version of @sawyerbfuller's subblk function would be ```sumblk = control.ss([], [], [], [[1, 0], [1, 1]])``` which avoids the creation of any additional states.  Note that this creates an "unnamed" system, since `StateSpace` systems do not have signal names associated with that.

* For systems with named signals (`InputOutputSystem`'s), this functionality is also available in the `interconnect` function "for free" by simply interconnecting multiple signals to the input of another block (they are automatically summed together and can also have optional gains assigned to them).  For example, the following code will create a feedback loop (without creating an explicit summing block for creating of the error term):
```
P = ct.LinearIOSystem(ct.tf2ss(ct.tf([1], [1, 1])), name='P')
C = ct.LinearIOSystem(ct.tf2ss(ct.tf([2], [1, 0])), name='C')
S = ct.interconnect([P, C],
    connections = [ ['P.u[0]', 'C.y[0]'], ['C.u[0]', '-P.y[0]'] ],
    inplist=['C.u[0]'], outlist=['P.y[0]'])
```

It would be possible to create a function that looks more like the MATLAB `sumblk` function, but if the only usage is for `InputOutputSystem`'s, then I am not sure whether it is needed (?).

k10blogger, 2021-01-10:
For named system, usage of sumblk as below is something that i feel is better readable:
```
 S1.InputName = 'r'
 S1.OutputName = 'ug1'
 S2.InputName = 'e'
 S2.OutputName = 'ug2'
 S3.InputName = 'ym'
 S3.OutputName = 'y'
 S4.InputName = 'u'
 S4.OutputName = 'ym'

 sum1 = sumblk('e','r','y','+-') % e = r - y
 sum2 = sumblk('u','ug1','ug2') % u = ug1 + ug2

connect(S1,S2,S4,S3,sum1,sum2,'r','ym')
```
It clarifies which systems shares input and which systems outputs is summed with what.

If it is possible to create a look alike `sumblk` with InputOutputSystems internal, that would work on provided named input formula then it would ease up the code a lot(?). 
Or if there is some other alternative where a formula can be used to depict system IO would be great to know as well.

murrayrm, 2021-01-10:
It's definitely possible to create a lookalike function for `sumblk' and would probably only take a few lines of code.  If you'd like to take a crack at it and submit a PR, I would encourage you to do so.  Rather than naming the function `sumblk`, we should probably call `summing_junction` to be more Pythonic about naming.  Then in the `matlab` module, we could set up an alias for `sumblk`.

An additional requirement (and interesting idea) would be to add an option in `interconnect` that automatically connects all signals that have the same name.  Combined with `summing junction` it could accomplish what you are describing.

closed 2021-01-26 by murrayrm