python-control/python-control #512: allow arrays of step, impulse responses for LTI systems

murrayrm, 2021-01-17:
The current implementation of step and impulse responses for MIMO systems requires the specification of an input channel, so that you always get a SIMO (or SISO) time response.  As pointed out in #453, this is not symmetric with the frequency responses, where you can index the input and output channels.  There is no reason this couldn't be done consistently for step and impulse responses by just iterating over the set of all input channels.  This would help make time and frequency responses more consistent.

Note that one inconsistency will remain: when you compute a forced response, you have to specify the particular input.  So in that case you can't index the input channel, only the output channel.

sawyerbfuller, 2021-01-18:
Agree about consistency argument. If input=None , could just return eg step
response to each input on a separate “row”.

Could do the same with forced response too, though specified input u data
is more likely in that function to be only of interest for the specified
input # (eg aileron vs throttle will have much different magnitudes).

S

murrayrm, 2021-01-18:
I have coded up this change, including `input=None` handling (separate row for each input).  Just waiting on merge of #511 before sending a PR.

This change breaks some of the unit test code that was implicitly assuming that `input=None` would select the first input.  But changes are minor and everything is pretty much backward compatible.  I suggest we implement this change into 0.9.0, though, since it may require modification of user code.

closed 2021-01-20 by murrayrm