python-control/python-control #513: Implement missing type conversions for block diagram algebra

murrayrm, 2021-01-18:
As described in issue #459 and PR #498, there are a number of type conversions for various overloaded operators that are not yet implemented.  A table of the current and desired operations can be found in [control/tests/type_conversion_test.py](https://github.com/python-control/python-control/blob/master/control/tests/type_conversion_test.py).

A few of the more useful ones:

* SISO `StateSpace` objects should be allowed to serve as a quotient under appropriate circumstances (so that you can write things like `T = P*C/(1 + P*C)` with state space objects, as you currently can with transfer functions.
* `FrequencyResponseData` and `InputOutputSystem` objects should be able to interact with floats and arrays.
* `InputOutputSystem` objects should be able to interact with `LTI` systems by converting the LTI system into a `LinearIOSystem`.
* `FrequencyResponseData` systems should be able to be combined with `StateSpace` and `TransferFunction` options by evaluating the state space system or transfer function at the FRD frequencies.

sawyerbfuller, 2021-02-11:
(this was originally its own issue, #541, but I copied the comment into here and closed that issue to keep discussion about its possible implementation in this more general thread.)

The following code causes a bug when it tries to raise an error: 
```python
import control as ct
import control.iosys as io
2*io.ss2io(ct.ss(1,1,1,1))
```

`raise NotImplemented("Scalar multiplication not yet implemented")`

This is because `InputOutputSystems` define various methods, e.g. `__mul__` that override those in `StateSpace` and some are not implemented yet. 

Options that come to mind if we want this to work better: 
1. Test for instances of `StateSpace` correctly and dispatch. This seems to be onerous and at odds with how inheritance ought to make things easier.
2. In the `__init__` method of `LinearIOSystem`, delete those methods
3. move those operations to NonLinearIOSystem (and override as necessary in `LinearIOSystem` when it makes sense to create a new `LinearICSystem`)

Related: many of the methods  like `__mul__` don't act on `self` and `other` but rather `sys1` and `sys2`. This might still work, but I think the first argument should be `self` according to Python convention.

murrayrm, 2021-12-30:
A number of these conversions have been implemented in #672.  Leaving this issue open since there are still a few cases that aren't handled (see table in [tests/type_conversion_test.py](https://github.com/python-control/python-control/blob/master/control/tests/type_conversion_test.py)).