python-control/python-control #516: Using interconnect/interconnected system

sawyerbfuller, 2021-01-22:
Trying to get the hang of creating interconnected systems, and hoping that `interconnect` can provide a more human-readable form of `connect`.

This is my question: If I'd like to create a simple feedback system like this: 
```
                    y
->0--->P(s)------>
  ^ -           |
  |    u        |
  ------C(s)<----
```
the following code gives the same system for both systems: 
```python
P = control.ss(control.tf(1, [1, 0]))
C = control.ss(control.tf(10, [1, 1]))
Pio = control.LinearIOSystem(P, inputs='u', outputs='y', name='P')
Cio = control.LinearIOSystem(C, inputs='y', outputs='u', name='C')
CL = control.interconnect((Pio, Cio), 
                                    connections = [['P.u', 'C.u'], ['C.y', '-P.y']],
                                    inplist='P.u', outlist='P.y')
display(control.feedback(P, C))
display(CL)
```

Is there any chance you could just, by correctly assigning names that match between systems as I have done here, interconnect automatically without needing the "connections" argument? Here, that doesnt seem to work (systems don't match anymore), but for some reason I had in my mind that it was possible. Am I just missing something? Thanks!

murrayrm, 2021-01-22:
This is not something that currently works (as you saw), but would be easy to implement via a keyword (`connect-by-name=True`?).  Basically we could go through and anytime the name of a signal in one system matches the name of a signal in another, we can connect things together.  If an output signal name appears multiple times, then you implicitly sum all outputs and attach to the inputs using that signal name.

One small issue is how to know the "sign" of the connection.  

MATLAB has a capability like this.  I'll need to look up how they handle the sign.

sawyerbfuller, 2021-01-23:
That must be it, I must have been thinking of Matlab.

 Suggested kwarg makes sense!

Not sure how they do negations but maybe one option is to include that in
signal name ‘-y’. Alternatively, just negate the system when defining it.

murrayrm, 2021-01-23:
Conceptually I like the idea of using -y, but not sure how that work work since sometimes you want the output (or input) to have a negative sign but sometimes you don't.  But there is only 1 signal name.  So, for example, if you had the system
```
--->0 ---> C(s) ---> P(s) ------>
    ^ -                      |
    |                        |
    -------------------------|
```
then the output from P needs to be positive at the output of the close loop system but negative at the input to C.

sawyerbfuller, 2021-01-23:
Makes sense & i agree that method would have a problem in a situation like you describe. the sign mioght have to be a property of the *connection* rather than the *signal*.  one fallback is to just rely on negative unity feedthroigh systems when necessary, but not as appealing because it would require more names.

murrayrm, 2021-01-23:
Another solution I just thought of is to use something like `sumblk` from issue #493.  If this was there, then you could do
```
P = ct.tf2io(control.tf(1, [1, 0]), inputs='u', outputs='y', name='P'))
C = ct.tf2io(control.tf(10, [1, 1]), inputs='e', outputs='u', name='C')
F = ct.summer(inputs=['r', '-y'], outputs='e')
S = control.interconnect((P, C, F), inplist='r', outlist='y')
```

murrayrm, 2021-01-23:
Code above works in PR #517

closed 2021-01-26 by murrayrm