python-control/python-control #526: Undesired model reduction

DarioTurchi, 2021-01-30:
Hello,
It seems to me that when I try to define a non-minimal state representation (e.g. with a state that can't be controlled or that doesn't have a dynamic) the system returned by the StateSpace class is automatically reduced, by eliminating such "non meaningful" states.
As example consider the following code:
`from control import StateSpace

a = [[0., 0., 0.], [0., 0., 0.], [0., 0., 0.]]
b = [[1, 0], [0, 0], [0, 1]]
c = [[1, 0, 0], [0, 1, 0], [0, 0, 1]]
d = [[0, 0], [0, 0], [0, 0]]
x = StateSpace(a,b,c,d)`

the obtained state representation is the following :
`>>> x
A = [[0. 0.]
 [0. 0.]]

B = [[1. 0.]
 [0. 1.]]

C = [[1. 0.]
 [0. 0.]
 [0. 1.]]

D = [[0. 0.]
 [0. 0.]
 [0. 0.]]
`

where the second state has been removed. Is it possible to not have this automatic model reduction (regardless of the fact that such state is not meaningful ) ? I couldn't find anything in the documentation

bnavigator, 2021-01-30:
Hello Dario,

this has recently been fixed in #509. Until it lands in a release, you can manually set `remove_useless_states=False`:

```
>>> from control.config import set_defaults
>>> set_defaults('statesp', remove_useless_states=False)
>>> x = StateSpace(a,b,c,d)
>>> x
StateSpace(array([[0., 0., 0.],
       [0., 0., 0.],
       [0., 0., 0.]]), array([[1., 0.],
       [0., 0.],
       [0., 1.]]), array([[1., 0., 0.],
       [0., 1., 0.],
       [0., 0., 1.]]), array([[0., 0.],
       [0., 0.],
       [0., 0.]]))

```

murrayrm, 2021-01-30:
Closing since this has been fixed (also duplicates #244).

closed 2021-01-30 by murrayrm

DarioTurchi, 2021-01-31:
I am really sorry to keep bothering you, but this is still not working for me. 
Do I need the package above a specific version ? Am I doing anything  stupid ?

I tried with both numpy array and lists for the definition of matrices A,B,C,D:
```
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from numpy import array
>>> from control import StateSpace
>>> from control.config import set_defaults
>>> a = array([[0., 0., 0.], [0., 0., 0.], [0., 0., 0.]])
>>> b = array([[1, 0], [0, 0], [0, 1]])
>>> c = array([[1, 0, 0], [0, 1, 0], [0, 0, 1]])
>>> d = array([[0, 0], [0, 0], [0, 0]])
>>> set_defaults('statesp', remove_useless_states=False)
>>> x = StateSpace(a,b,c,d)
>>> x
A = [[0. 0.]
 [0. 0.]]

B = [[1. 0.]
 [0. 1.]]

C = [[1. 0.]
 [0. 0.]
 [0. 1.]]

D = [[0. 0.]
 [0. 0.]
 [0. 0.]]

>>> 
```

```
Python 3.8.5 (default, Jul 28 2020, 12:59:40) 
[GCC 9.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import control as ctr
>>> ctr.config.set_defaults('statesp', remove_useless_states=False)
>>> a = [[0., 0., 0.], [0., 0., 0.], [0., 0., 0.]]
>>> b = [[1, 0], [0, 0], [0, 1]]
>>> c = [[1, 0, 0], [0, 1, 0], [0, 0, 1]]
>>> d = [[0, 0], [0, 0], [0, 0]]
>>> x = ctr.StateSpace(a,b,c,d)
>>> x
A = [[0. 0.]
 [0. 0.]]

B = [[1. 0.]
 [0. 1.]]

C = [[1. 0.]
 [0. 0.]
 [0. 1.]]

D = [[0. 0.]
 [0. 0.]
 [0. 0.]]
```

sawyerbfuller, 2021-01-31:
Try remove_useless=False?

(Leave out ‘states’)

the constructor should raise an error if it receives an un-recognized
keyword like this.
-- 
Sawyer Fuller
Assistant Professor of Mechanical Engineering
Adjunct, Paul G Allen School of Computer Science
University of Washington
http://faculty.washington.edu/~minster/

(Typed with my thumbs)

murrayrm, 2021-01-31:
Yes, in v0.8.3 the keyword was called `remove_useless`, so you have to use
```
x = ctr.StateSpace(a,b,c,d, remove_useless=False)
```
Note also that v0.8.3 did not use the `config` module to set the default value, so calling `set_defaults` won't work in that version.

v0.8.4 changes the name of the keyword to `remove_useless_states` and sets the default value to `True` (but you can use `set_defaults` to override).

v0.9.0 (upcoming) sets the default value to `False`.

DarioTurchi, 2021-01-31:
Many thanks