python-control/python-control #530: new iosys.interconnect is great

sawyerbfuller, 2021-02-03:
The new naming functionality in `iosys.interconnect` function (https://github.com/python-control/python-control/blob/73f65dfe1dbf3ca9431a815769da42c52feb6169/control/iosys.py#L1850) has allowed me to create some very sophisticated systems much more easily than with `append` and `connect`. So far, no bugs that I have found. Very nice!  

a few remarks on usage: 
* the `inplist` and `outlist` arguments are confusing and tricky. When you are writing code, its easy to forget shortened argument names like this: was it `outplist` or `outlist`? Also, they seem to exist to serve as a means to provide new names to the overall inputs and outputs. Is that a necessary feature? (might have to do with how system was set up for explicit interconnection) Possible remove them and force the overall system to have the input names and output names as constituent systems, and instead just provide `inputs` and `outputs` kwargs for `interconnect`? 
* auto-summing and auto-splitting, as suggested here: https://github.com/python-control/python-control/blob/master/doc/iosys.rst#automatic-connections-using-signal-names doesn't seem to work. For example, when I try to send the same signal `u` to both `sys1` and `sys2` in the following code a `ValueError: signal u is not unique` is raised:  
```python
sys1 = ct.tf(1, [10, 1])
sys2 = ct.tf(1, [2, 1])
summer = ct.iosys.summing_junction(['y1', 'y2'], 'y')
sys1 = ct.iosys.tf2io(sys1, inputs='u', outputs='y1')
sys2 = ct.iosys.tf2io(sys2, inputs='u', outputs='y2')
parallel = ct.interconnect((summer, sys1, sys2), inplist='u', outlist='y')
```
* similarly for auto-summing. the following raises `ValueError: signal y is not unique`:
```python
splitter = ct.ss2io(ct.ss([], [], [], [[1],[1]]), inputs='u', outputs=('u1', 'u2'))
sys1 = ct.iosys.tf2io(sys1, inputs='u1', outputs='y')
sys2 = ct.iosys.tf2io(sys2, inputs='u2', outputs='y')
parallel = ct.interconnect((splitter, sys1, sys2), inplist='u', outlist='y')
```

I don't know how hard these are to fix, it depends on how the underlying interconnection code works. One option would be to just modify the docs to remove mention of auto splitting and summing and add a helper splitter functoin: 
```python
def splitter(inputs, outputs):
    D = np.ones((len(outputs),1))
    C = np.zeros_like(D)
    return io.ss2io(ct.ss(0, 0, C, D), inputs=inputs, outputs=outputs)
```

murrayrm, 2021-02-04:
The lack of autosumming is a bug and I can fix that one pretty quickly.  Adding a splitter function might also be useful, so I'll probably do that at the same time.

For `inplist`/`inputs` and `outlist`/`outputs`: as @sawyerbfuller points out, there are two things that a user might want to do:

* Choose which signals are inputs and outputs (from the set of signals inside the system, including possible sums of those signals)
* Name the signals

The idea is that `inplist`/`outlist` specifies the signals (including allowing summation and splitting) and `inputs`/`outputs` names them.  Note that if you aren't using the autoconnect functionality then signal names are not unique without the system name prepended (so there might be multiple 'u[0]' signals, but only one 'sys[0].u[0]' signal) => not sure we want to just use the underlying signal name (it could get ugly if you have a couple of levels of nesting, so that your overall system output becomes `sys[0].sys[1].y[0]` for example).  It could also get ugly with implicit summing (an output could become 'sys[0].y[0]+sys[1].y[1]' for example).

sawyerbfuller, 2021-02-04:
> The lack of autosumming is a bug and I can fix that one pretty quickly. Adding a splitter function might also be useful, so I'll probably do that at the same time.

<thumbs up>

> For `inplist`/`inputs` and `outlist`/`outputs`: as @sawyerbfuller points out, there are two things that a user might want to do:
> 
>     * Choose which signals are inputs and outputs (from the set of signals inside the system, including possible sums of those signals)
> 
>     * Name the signals
>
Makes sense.
> The idea is that `inplist`/`outlist` specifies the signals (including allowing summation and splitting) and `inputs`/`outputs` names them. Note that if you aren't using the autoconnect functionality [...]

For the output of an autosumming version of the summing-parallel-systems example above, it makes sense that you would not want the output name be  `'y'` if it is actually `sys1.y + sys2.y` (looks like in that case Matlab forces you to create a summing block with a new name).

I think I understand now. I am curious to see how it will work in practice with autosumming.

sawyerbfuller, 2021-02-07:
I’ve reread the text of this issue, and I think the name change is fair enough : )

closed 2021-02-09 by sawyerbfuller