python-control/python-control #538: Freqplot improvements introduced warning messages in tests

murrayrm, 2021-02-08:
It looks like PR #522 introduced some warning messages in our GitHub Action tests:
```
control/tests/freqresp_test.py::test_discrete[tf-0.1]
control/tests/freqresp_test.py::test_discrete[tf-0.1]
control/tests/freqresp_test.py::test_discrete[tf-True]
control/tests/freqresp_test.py::test_discrete[tf-True]
  /home/runner/work/python-control/python-control/control/xferfcn.py:310: RuntimeWarning: divide by zero encountered in true_divide
    out[i][j] = (polyval(self.num[i][j], x) /

control/tests/freqresp_test.py::test_discrete[tf-0.1]
control/tests/freqresp_test.py::test_discrete[tf-0.1]
control/tests/freqresp_test.py::test_discrete[tf-True]
control/tests/freqresp_test.py::test_discrete[tf-True]
  /home/runner/work/python-control/python-control/control/xferfcn.py:310: RuntimeWarning: invalid value encountered in true_divide
    out[i][j] = (polyval(self.num[i][j], x) /
```
I'm not quite sure what is generating the warnings, but it would be good to get rid of them if we can.

Test results here: https://github.com/python-control/python-control/pull/522/checks

@sawyerbfuller Any ideas?

sawyerbfuller, 2021-02-08:
Hadn’t seen that. Not at my computer so I can’t yet confirm, but I think that if those are all DT systems, a possible explanation is that I changed the code to evaluate all the way up to and including the nyquist frequency. A pole at -1 might cause such a warning.

murrayrm, 2021-02-08:
It looks like they are all discrete time systems, so that's probably it.

I'm working on some code for issue #532 and will be adding some flags to turn off warnings, so that is one possible fix.  But probably worth thinking about what the right approach is in terms of the evaluating/plotting the frequency response at those values.

sawyerbfuller, 2021-02-09:
Yep, confirmed the system has a pole at -1 and that's what causes a warning when running bode. Thinking about what to do about it. Bode plot still works but it has a gap in the line near the nyquist plot (from the nan's).

closed 2021-02-12 by murrayrm