python-control/python-control #551: Numpy automatically casting Complex to Float

Bruno-SW, 2021-02-24:
It's been a while since i've used this awesome work you guys did.
Tried some simple things today and couldn't get it to work

I believe numpy got updated and messed some things up

Given the following code:
```
import control as ct
import numpy as np

# Define Poles and get it's coeficients
poles = np.array([np.complex(-7.7, 6.4), np.complex(7.7, 6.4)]) # Expecting to get these poles when G.pole() is called
coefs = np.poly(poles)

# Define the Transfer Function
num = [1]
den = coefs
G = ct.tf(num, den)

# Get the poles
G.pole()
```

i get these warnings:

`
/usr/local/lib/python3.7/dist-packages/control/xferfcn.py:957: ComplexWarning: Casting complex values to real discards the imaginary part den[j, :maxindex+1] = poly(poles[j])
`
`
/usr/local/lib/python3.7/dist-packages/control/xferfcn.py:987: ComplexWarning: Casting complex values to real discards the imaginary part num[i, j, maxindex+1-len(numpoly):maxindex+1] = numpoly
array([-10.0124922,  10.0124922])
`

Doing a few tests i found out that

```
poles = np.array([np.complex(-7.7, 6.4), np.complex(7.7, 6.4)])

maxindex = 2
den = np.zeros((1, maxindex + 1), dtype=np.float)
den[0, :maxindex + 1] = np.poly(poles)
```
`/usr/local/lib/python3.7/dist-packages/ipykernel_launcher.py:5: ComplexWarning: Casting complex values to real discards the imaginary part`
will results in the same warnings, but changing dtype to np.complex it works correctly
```
poles = np.array([np.complex(-7.7, 6.4), np.complex(7.7, 6.4)])

maxindex = 2
den = np.zeros((1, maxindex + 1), dtype=np.complex)
den[0, :maxindex + 1] = np.poly(poles)
den
```
`array([   1.   +0.j ,    0.  -12.8j, -100.25 +0.j ])`

I belive changing the num and den dtype to np.complex from xferfcn.py file should solve the issue (did not look up for similar issues/solutions though)
```
938        den = zeros((self.inputs, maxindex + 1), dtype=np.complex)
939        num = zeros((max(1, self.outputs, self.inputs),
940                     max(1, self.outputs, self.inputs),
941                     maxindex + 1),
942                    dtype=np.complex)
```

murrayrm, 2021-02-25:
It looks like you are trying to put poles at `-7.7 + 6.4j` and `+7.7 + 6.4j`, which gives a transfer function with complex coefficients.  This isn't yet supported (but see PR #484 for some initial steps).

If you do complex conjugate poles, everything looks OK:
```
import control as ct
import numpy as np

# Define Poles and get it's coeficients
poles = np.array([complex(-7.7, 6.4), complex(-7.7, -6.4)]) # Expecting to get these poles when G.pole() is called
coefs = np.poly(poles)

# Define the Transfer Function
num = [1]
den = coefs
G = ct.tf(num, den)

# Get the poles
G.pole()
```

closed 2021-02-25 by murrayrm

Bruno-SW, 2021-02-25:
I see... i messed up, thanks for the feedback :)