python-control/python-control #554: step_info overshoot error

juanodecc, 2021-03-02:
Hi, the overshoot calculate by step_info function is wrong for this transfer function:

```
import control as ctrl #control 0.8.3
import matplotlib.pyplot as plt #matplotlib 3.2.2

transferFunction = ctrl.tf([0.6287,37.74,774.4,5797,7515],[1.629,45.74,788.4,5809,7515])

time, out = ctrl.step_response(transferFunction)

plt.figure()
plt.grid()
plt.plot(time, out)
```
[image]
```
ctrl.step_info(transferFunction)
```

> {'RiseTime': 0.051990934709959705,
 'SettlingTime': 0.4289252113571676,
 'SettlingMin': 0.9121688025464874,
 'SettlingMax': 1.1826717168233585,
 'Overshoot': 29.748622946474416,
 'Undershoot': 0.3859422958870473,
 'Peak': 1.1826717168233585,
 'PeakTime': 0.13430991466739592,
 'SteadyStateValue': 0.9999984629917944}

bnavigator, 2021-03-02:
Thanks for bringing up the issue @juanodecc. I can confirm, although the plot is smoother with the new timevector calculation in 0.8.4 (#420), that the `Overshoot` value is still not what you would expect.

[image]

https://github.com/python-control/python-control/blob/8cba9493ab1efa54577846a421fe3ed7b2893339/control/timeresp.py#L817

`yout[0]` here is 0.386 instead of 0. Not sure, why this is referenced here. The `step_response()` of a system without any further specification of initial values should start from default initial value of 0.

@sawyerbfuller, do you think the response vector calculation should be adjusted or just the calculation of `Overshoot`?

sawyerbfuller, 2021-03-04:
This is a good catch. `out[0]` is not zero because this is not a strictly proper transfer function: the order of the numerator and denominator are the same. 

My take on it is that the overshoot calculation has a bug that neglects this case. It should not depend on the value of `yout[0]`.

closed 2021-03-08 by bnavigator