python-control/python-control #557: MIMO Transfer Function

CtrlAlt-Delete, 2021-03-10:
Let's say I have a MIMO system where the numerator is
`NUMv = [[1], [2]]`
and the denomenator
`DENv = [[1, 2], [3, 4]]`
and I try to make a transfer function by
`Hv = tf(NUMv, DENv)`
This is how I would set it up according to how I've understood the docs. 

However, I get this error message:
**"TypeError: The numerator and denominator inputs must be scalars or vectors (for SISO), or lists of lists of vectors (for SISO or MIMO)."**

What am I missing? What triggers the TypeError? I've tried setting it up the same way in Matlab, which was successful, but I need it to work in Python as well.

Thanks in advance!

ilayn, 2021-03-10:
Nest them one more pair of `[]` to make it list of lists as if your are doing a 3d array. Currently you are trying to build 
```python
      [1]
      [2]
-----------
    [ 1 2 ]
    [ 3 4 ]
```

So you need 

```python
Numv = [[[1], [2]]]
Denv = [[[1,2], [3,4]]]
```

murrayrm, 2021-03-10:
@ilayn beat me to it, but here is a bit more detail.

As you have it written, the meaning is ambiguous: are you trying to create a 2-input, 1-output system or a 1-input, 2-output system?

For 2-input, 1-output, use
```
H = control.tf([[ [1], [2] ]], [[ [1, 2], [3, 4] ]] )
```
For 1-input, 2-output, use
```
H = control.tf([ [[1]], [[2]] ], [ [[1, 2]], [[3, 4]] ])
```

CtrlAlt-Delete, 2021-03-11:
Thank you! 

I was trying to create a 2-input 1-output system. 

You solution worked like a charm!

closed 2021-03-11 by CtrlAlt-Delete