python-control/python-control #559: Question: feedback loop for temperature control via peltier

kratsg, 2021-03-11:
This looks like a very nice (and well-developed) package. I haven't done the control systems through matlab (which a lot seems to be referenced) but I'd like to get started, maybe using this package to describe a system that provides feedback on the control of a peltier element (via modifying the voltage on a power supply).

I'm not entirely sure where to get started, but hoping I could get some tips or be pointed along the right path here. Nominally, the functionality would be that the current temperature, reference (goal) temperature, and current voltage as input -- with output being the new voltage to set to. [Perhaps an additional input would be a time-delta between measurements]. Thanks!

bnavigator, 2021-03-11:
Hi,

of course there is the [package documentation](https://python-control.readthedocs.io/en/0.8.4/).
You can also start by comparing the examples given in [FBS](https://fbswiki.org/wiki/index.php/Feedback_Systems:_An_Introduction_for_Scientists_and_Engineers) with those [implemented for python-control](https://github.com/python-control/python-control/tree/master/examples), like the basic cruise control.

kratsg, 2021-03-12:
> the basic cruise control.

Sorry, I did see the cruise control example, but that doesn't seem basic? Is this package not meant to be used for something like a simple PID controller? Because it would be really helpful to show how to use it for a simple application like that.

Personally, that's a very long example to try and digest, and I'm not even sure I can clearly formulate a set of equations in my case that describe the temperature flow of a complicated system.

closed 2021-03-12 by bnavigator