python-control/python-control #560: Separate Discussions from Issues

bnavigator, 2021-03-11:
Every now and then there is a question about using python-control, which is not a bug, but posted to the GitHub issue page. There is the mailing list. But many people like to post questions on web sites rather than using old fashioned e-mail. How about we open the ["Discussions"](https://docs.github.com/en/discussions) feature of GitHub for this kind of questions?

murrayrm, 2021-03-12:
Done.  Leaving this open to discuss how we want to use it.

Seems clear that questions and help on using the package would go into discussions.  Anything else?  I guess feature requests still come here?

Also, is there a way to move something from Issues to Discussions?

bnavigator, 2021-03-12:
It seems possible: https://docs.github.com/en/discussions/managing-discussions-for-your-community/moderating-discussions#converting-an-issue-to-a-discussion

murrayrm, 2021-03-12:
Cool. Looks like this is now working, so closing this issue.  Thanks for pointing out this option @bnavigator!

closed 2021-03-12 by murrayrm