python-control/python-control #573: control.matlab.freqresp is inconsistent with MATLAB

murrayrm, 2021-03-15:
As pointed out in issue #550, the `freqresp` function returns magnitude, phase, and frequency instead of the complex frequency response.  This is OK for the main package (since you can just evaluate the system itself to get the complex frequency response), but it is inconsistent with [MATLAB's freqresp function](https://www.mathworks.com/help/control/ref/lti.freqresp.html).  At the least, `control.matlab.freqresp` should be consistent with MATLAB.

See discussion in PR #449 and issue #434.

sawyerbfuller, 2021-03-16:
Looking through the code, it looks like the `freqresp` *function*, which is in the `lti` module, gets imported correctly in the matlab module.  (Am not at my computer so haven’t tested this yet). The `sys.freqresp` *method*, on the other hand, is deprecated in favor of`sys.frequency_response`. So I think the problem might only be that it returns the results in the wrong form (mag and phase rather than complex).

murrayrm, 2021-03-16:
Agh.  I must have been looking at some different version.  I think the problem is not that the function is missing, but that it is inconsistent with MATLAB.  Re-editing title and description.  Sorry for the confusion.

JinmingChe, 2021-03-24:
Hi, I update the control to version 0.9.0. LTI system can be callable. But I want to caculate the freqresp of a FRD system to get the complex frequency response. If there any solution?

murrayrm, 2021-03-24:
Use `H.eval(omega*1j)` or `H.eval(omega)` where `H` is an FRD function.  If you want to evaluate at frequencies other than those that define the frequency response, you need to create a smooth FRD object (use `frd(d, w, smooth=True)`).