python-control/python-control #574: step_info error for MIMO system

juanodecc, 2021-03-16:
I try with this MIMO system, the function broke at line 614. The step_info function convert the system to SIMO, but cannot calculate the step response characteristic parameters.

```
from control import ss step_response step_info

A = [[0.68, -0.34], [0.34, 0.68]];
B = [[0.18, -0.05], [0.04, 0.11]];
C = [[0, -1.53], [-1.12, -1.10]];
D = [[0, 0], [0.06, -0.37]];
m_sys = ss(A,B,C,D,0.2);
mt,my=step_response(m_sys)

step_info(m_sys)

```

>   File ".../timeresp.py", line 614, in step_info
>     if((yout[i] <= inf_margin) | (yout[i] >= sup_margin)):
> 
>  IndexError: index 24 is out of bounds for axis 0 with size 2
>

bnavigator, 2021-03-16:
Why line 614? Current master has that line at 807, Your own PR #567 duplicates it in 807 and 817.

Can confirm with current master. The problem is the indexing of yout:

```
[ben@skylab:~/src/python-control]% python3                                                                                                                       [0]
Python 3.8.8 (default, Feb 19 2021, 16:53:21) [GCC] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from control import ss, step_response, step_info
>>> 
>>> A = [[0.68, -0.34], [0.34, 0.68]];
>>> B = [[0.18, -0.05], [0.04, 0.11]];
>>> C = [[0, -1.53], [-1.12, -1.10]];
>>> D = [[0, 0], [0.06, -0.37]];
>>> m_sys = ss(A,B,C,D,0.2);
>>> mt,my=step_response(m_sys)
>>> mt.shape
(25,)
>>> my.shape
(2, 2, 25)
>>> step_info(m_sys)
/home/ben/src/python-control/control/statesp.py:1549: UserWarning: Converting MIMO system to SIMO system. Only input 0 is used.
  warn("Converting MIMO system to SIMO system. "
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/ben/src/python-control/control/timeresp.py", line 807, in step_info
    if((yout[i] <= inf_margin) | (yout[i] >= sup_margin)):
IndexError: index 24 is out of bounds for axis 0 with size 2
```

bnavigator, 2021-03-16:
Why not adjust #567 to handle MIMO systems the way [MATLAB's `stepinfo`](https://mathworks.com/help/control/ref/lti.stepinfo.html#mw_974c4f92-9fca-4f0e-bd68-032d21335622) does:

> For a MIMO system, stepinfo returns a structure array in which each entry contains the response characteristics of the corresponding I/O channel of the system. For this example, use a two-output, two-input discrete-time system. Compute the step-response characteristics.

juanodecc, 2021-03-16:
> Why line 614? Current master has that line at 807, Your own PR #567 duplicates it in 807 and 817.
>
> Can confirm with current master. The problem is the indexing of yout:
>
> ``` [...]

yes my bad, is line 614 for version 8.4

juanodecc, 2021-03-16:
> Why not adjust #567 to handle MIMO systems the way [MATLAB's `stepinfo`](https://mathworks.com/help/control/ref/lti.stepinfo.html#mw_974c4f92-9fca-4f0e-bd68-032d21335622) does:

Yes, I am trying to fix it, but I commit a partial solution for now. That improve the performance of this function

juanodecc, 2021-03-16:
> Why not adjust #567 to handle MIMO systems the way [MATLAB's `stepinfo`](https://mathworks.com/help/control/ref/lti.stepinfo.html#mw_974c4f92-9fca-4f0e-bd68-032d21335622) does:

step_response have limitations, that function can solve only SIMO systems. My idea is convert the step_info into two functions: step_info4data (t, y, yfinal) for multidimentional, time and output parameters, in the other hand, keep a step_info(sys) as is, but that function can call step_info4data () with the proper input vector for MIMO systems. This can allow use the step_info4data function with  time vectors and output vectors imported from externals simulation or system stimulation

bnavigator, 2021-03-16:
I might be missing something, but the current `step_response` can handle MIMO. See `my.shape`.

PR #577 (2021-03-20): enhance step_info to MIMO and time series of response data
Enhance the `step_info()` function to return a 2D list of info dicts for MIMO systems.

Cleaned up the code a little.

Fixes #574

WIP: needs unit tests for MIMO
closed 2021-03-20 by murrayrm