python-control/python-control #591: peak value and peak time calculus detail

juanodecc, 2021-03-25:
Hi @bnavigator for asymptotic systems and the absolute undershoot value is less than yfinal the peak value and peak time calculated are different for different time vector. I explain it using this example:

```

from control import tf, step_response, step_info, ss
import matplotlib.pyplot as plt
import numpy as np

transfer_function = tf( [3.32, 0, -162.8], [1, 24.56, 186.5, 457.8, 116.2] )

step_info(transfer_function)
step_info(transfer_function,T=np.linspace(0,32,2000))
step_info(transfer_function,T=np.linspace(0,40,2000))
step_info(transfer_function,T=np.linspace(0,100,10000))

```
I think, we can solve it, doing:

```
                steady_state_value_mod = np.abs(steady_state_value)
                if peak_time == T[-1] and  peak_value < steady_state_value_mod: # Asymptotic system
                    peak_value = steady_state_value_mod
                    peak_time = np.Inf 

```

I check with this other example

```
from control import tf, step_response, step_info, ss
import matplotlib.pyplot as plt
import numpy as np

G1 = tf( [-1, 1], [1, 1] )

time, yout = step_response(G1)

fig, ax = plt.subplots(1, 1)
ax.plot(time, yout)
plt.show()

info1=step_info(G1)
print(info1)

G2 = tf( [-0.9, 1], [1, 1] )

time, yout = step_response(G2)

fig, ax = plt.subplots(1, 1)
ax.plot(time, yout)
plt.show()

info2=step_info(G2)
print(info2)
```

bnavigator, 2021-03-25:
Sounds good to me. How about you open a PR? Please add the examples with reference values to the list of tested systems for `test_step_info()`.

BTW, you don't have to @ me. The team of developers is a bit larger and anyone can review your suggestions.

juanodecc, 2021-03-26:
Sorry I @ you because you are working in this function. I think this is an important modification, I'm not sure if this concept is right. If anybody else want to say something, are welcome. I will do the PR, and we can discus it thear.