python-control/python-control #592: Singular values plot missing

forgi86, 2021-03-25:
Among the frequency-domain analysis functions, a singular values plot is missing in the current version 
of python-control. Is there any plan to include it? I think it is very common and useful for MIMO system analysis.

bnavigator, 2021-03-25:
I think it depends on what exactly you want to see implemented.

A simple wrapper of feeding MIMO transfer functions or `FrequencyResponseData` objects into `numpy.linalg.svd` and then calling a Matplotlib plot function does not really add any value over calling those functions directly, does it?

forgi86, 2021-03-25:
The matlab equivalent (function sigma) does this plot on an automatically adjusted logarithmic frequency scale, with singular values in dB, etc. Eventually it should look like a bode plot, but with singular values instead. Sure you can build this plot from scratch, but it takes some time boilerplate code. Today for instance I needed it while working in Python, but I was lazy so I switched to the commercial software :) So I think wrapping it up in a nice way would add some value!

bnavigator, 2021-03-26:
Sounds reasonable. You are always welcome to submit a PR.

forgi86, 2021-03-26:
OK maybe I'll try to do it when I have time!

closed 2021-04-29 by bnavigator