python-control/python-control #594: squeeze: return shape of the frequency response.

bnavigator, 2021-03-28:
Something strange about the return shape of the frequency response. For a square 2x2 system:

sys(np.array(0.0), squeeze=False).shape
(2, 2)

sys(0.0, squeeze=False).shape
(2, 2, 1)

_Originally posted by @forgi86 in https://github.com/python-control/python-control/issues/593#issuecomment-808929799_

bnavigator, 2021-03-28:
BTW, omega has to be complex.

```
>>> import numpy as np
>>> from control import rss
>>> sys = rss(2,2,2)
>>> sys(np.array(0.0j), squeeze=False).shape
(2, 2, 1)
>>> sys(0.0j, squeeze=False).shape
(2, 2)
```

Now I am even more confused and will stop for this Sunday evening.

murrayrm, 2021-03-28:
I think this behavior is expected.  If you call an LTI system with a list of frequencies, you get back a list of responses.

murrayrm, 2021-03-28:
Sorry, accidently closed.

forgi86, 2021-03-29:
@bnavigator as far as I can see looks like you can pass real/complex/imaginary argument. I would expect the output to be the Laplace/Z transform for ct/dt in general, but haven't really tested!

PR #595 (2021-04-01): lti squeeze: ndarray.ndim == 0 is also a scalar
Fixes #594
closed 2021-04-01 by bnavigator