python-control/python-control #604: __call__ and other new methods not described in sphinx docs

sawyerbfuller, 2021-04-02:
I wanted to update docs to include new methods like __call__ and dynamics, because they currently don't appear, e.g. in : https://python-control.readthedocs.io/en/0.9.0/generated/control.TransferFunction.html#control.TransferFunction

I think this might be somehow autogenerated, but I can't figure out how. The helpful "edit on github" button at top right gives a 404. Can somebody point me to how to make sure those methods appear in the next docs release?

murrayrm, 2021-04-02:
They are generated using Sphinx, with the source files sitting in `doc/`.  Some information is pulled from docstrings, but you can also put in documentation separately.

murrayrm, 2021-04-03:
Note: `dynamics` appears correctly in [StateSpace documentation](https://python-control.readthedocs.io/en/0.9.0/generated/control.StateSpace.html).

For `__call__`, it looks like we may have to turn on sphinx processing of special-members.  See [autodoc documentation](https://www.sphinx-doc.org/en/master/usage/extensions/autodoc.html).  I tried playing with this, but couldn't get it to work.  Will take more investigation.

If you have sphinx installed locally, you can run `make html` in `doc/` and then open `_build/html/index.html` to see what the documentation will look like.

sawyerbfuller, 2021-04-06:
@murrayrm I see, you're right, ``dynamics`` does appear in the sstatespace doc, just not in the transferFunction doc, as you would expect. I'll take a look at the link you sent for autodoc. 

Another issue I noticed is that the attribues are neither correct in name (`inputs` and `outputs` are included rather than `ninputs` and `noutputs`, and `dt` is not included), and there is no information about them. I'll follow that link and see if I can find what to do there.

closed 2021-06-16 by murrayrm