python-control/python-control #613: Nyquist plot with imaginary poles, semi-circles only working after 'trick'

kevindekemele, 2021-04-22:
I'm plotting the Nyquist diagram of a transfer function which has poles on the imaginary axis:

```

import control
import matplotlib.pyplot as plt
import numpy as np

K = 1

Gp =  control.tf([1], [1,1,0,0,0]) 
Gc = control.tf([K], [1]) 

control.nyquist_plot(Gp*Gc,plot=True,omega_num=1000)
plt.show()

```
Which results in the following incorrect plot, that does not show infinte semicircles, but something incorrect:
[image]

Also using `indent_direction` and `indent_radius` did not solve this.
The trick I found that makes it work is if I add `omega_limits=[0.001,10000]` to nyquist_plot:

[image]

Can you please fix such that the incorrect 'line' does not appear?

bnavigator, 2021-04-24:
Yes, the algorithm to determine the default omega vector for the system starts with a lower frequency of 0.1, because the pole of the system is at -1.

https://github.com/python-control/python-control/blob/e81f648431e2a9f2cdf583702522bc2af0e1c360/control/freqplot.py#L1088-L1092

The "trick" to manually define the omega limits is the correct way to override the default omega vector determination.

kevindekemele, 2021-04-25:
Perhaps if there is a pole at the origin, the lower limit of the omega vector should be below the indent_radius ?

closed 2021-06-16 by murrayrm