python-control/python-control #623: I'm having problems with the `ax` parameters of the `rlocus` function

jolenscki, 2021-05-22:
I was trying today to plot the root locus of a transfer function, but wanted to pass a specific axes object to it, so I tried to use the `ax` parameter of the `rlocus` function:

```
from control import *
import matplotlib.pyplot as plt

s = tf([1, 0], 1)
G = 1/(s*(s+1))

fig, ax = plt.subplots(dpi=200)
rlocus(G, ax=ax)
```

But when doing this, the following error was raised:

```
---------------------------------------------------------------------------
UnboundLocalError                         Traceback (most recent call last)
<ipython-input-1-80498c4b1a54> in <module>
      6 
      7 fig, ax = plt.subplots(dpi=200)
----> 8 rlocus(G, ax=ax)

~\anaconda3\lib\site-packages\control\rlocus.py in root_locus(sys, kvect, xlim, ylim, plotstr, plot, print_gain, grid, ax, **kwargs)
    173 
    174         if print_gain and not sisotool:
--> 175             fig.canvas.mpl_connect(
    176                 'button_release_event',
    177                 partial(_RLClickDispatcher, sys=sys, fig=fig,

UnboundLocalError: local variable 'fig' referenced before assignment
```

Control version: 0.9.0

sawyerbfuller, 2021-05-22:
Looks like you caught a bug, thanks! I think when we implemented this we had what you intended in mind but didn’t get around to testing it.

billtubbs, 2021-08-09:
It's not too difficult to fix.  I think we just need to unindent lines 171 and 172 by one tab:

https://github.com/python-control/python-control/blob/283f5e7480a7c39f0e11abe63e3c1ecd8b5d8911/control/rlocus.py#L169-L172

bnavigator, 2021-08-09:
If you manage to also fix #634 while you are at it, it would be a blast. Not sure if it is worth to be done before/without implementing the new plotting paradigm, proposed by you in https://github.com/python-control/python-control/issues/65#issuecomment-894927338

billtubbs, 2021-08-09:
I'll have a look at #634.  But I'm waiting for further input on the new plotting paradigm. That will require more thinking and discussion and probably take a while to figure out a way forward. There might be better ways to do it. I'm happy to try fixing the immediate bugs on the existing functions when I next get the chance.

closed 2021-12-26 by bnavigator