python-control/python-control #624: lsim warning for TF systems

mvaccariunipi, 2021-05-27:
Dear all,
I have just installed control 0.9.0 and when using lsim (or forced_response) on a transfer function system I got the following warning: "return_x specified for a transfer function system. Internal conversion to state space used; results may meaningless."

Can I skip this warning? Is there another function for simulating transfer function systems that does not require a conversion to state space? 
I cannot figure out if the warning is something I should worry about or not. 

Thanks,

Marco from SIPPY teams

The example tested is the following:

from __future__ import division
from past.utils import old_div
from sippy import functionset as fset
from sippy import *
import numpy as np
import control.matlab as cnt

## TEST OPTIMIZATION-BASED IDENTIFICATION METHODS for GENERAL INPUT-OUTPUT MODEL

# Define sampling time and Time vector
sampling_time = 1.                                  # [s]
end_time = 400                                      # [s]
npts = int(old_div(end_time, sampling_time)) + 1
Time = np.linspace(0, end_time, npts)

# Define Generalize Binary Sequence as input signal 
switch_probability = 0.08  # [0..1]
[Usim,_,_] = fset.GBN_seq(npts, switch_probability, Range = [-1, 1])

# Define white noise as noise signal
white_noise_variance = [0.01]
e_t = fset.white_noise_var(Usim.size, white_noise_variance)[0]

# ## Define the system (BOX-JENKINS model)

# ### Numerator of noise transfer function has two roots: nc = 2

# ## Define the system (BOX-JENKINS model)

# ### Denominator between input and noise transfer functions has 4 roots: nf = 4

DEN = [1., -2.21, 1.7494, -0.584256, 0.0684029, 0., 0., 0., 0., 0., 0., 0., 0., 0., 0.]

# ### Numerator of input transfer function has 3 roots: nb = 3

NUM = [1.5, -2.07, 1.3146]

# ### Define transfer functions
g_sample = cnt.tf(NUM, DEN, sampling_time)

Y1, Time, Xsim = cnt.lsim(g_sample, Usim, Time)

sawyerbfuller, 2021-05-27:
You can ignore that warning if you don't intend to make use of x. x is the "internal state" of the system, and depends on the specific realization of your state-space representation of the transfer function you provided.

mvaccariunipi, 2021-05-27:
Hi @ sawyerbfuller and thanks for the prompt answer. Is there a way that to not show the warning? Otherwise I'll just ignore it. 

Thanks again,

Marco

sawyerbfuller, 2021-05-27:
Hi Marco, 

one way is to use the more pythonic forced_response, which gives you more control, and in particular, I believe you can ask it to not give you x.

closed 2021-05-27 by bnavigator