python-control/python-control #634: Two 'sgrid' functions when we only need one?

murrayrm, 2021-06-16:
As pointed out in PR #617, there are now two `sgrid` functions in the package: `grid.py:sgrid` and `rlocus.py:_sgrid_func`.  It looks like `_sgrid_func` was removed in PR #193 (discrete omega-damping plot and tweaks by Sup3rGeo), but was re-introduced in PR #209 (sisotool and dynamic root locus zoom by icam0).

bnavigator, 2021-06-16:
This should be fixed together with #632 and #623

billtubbs, 2021-08-15:
I'm planning to look at this tomorrow.  Does anyone know why `rlocus.py` calls its own local [`_sgrid_func(fig=None, zeta=None, wn=None)`](https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/rlocus.py#L642) instead of [`grid.sgrid()`](https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/grid.py#L68) (which is already imported into the module but is not used).  There might be a reason why someone did this...

Something to do with the `zeta` and `wn` probably?

bnavigator, 2021-08-15:
> I'm planning to look at this tomorrow. Does anyone know why `rlocus.py` calls its own local [`_sgrid_func(fig=None, zeta=None, [...]

It is basically a result of two PRs which were started indepently and merged after some time: https://github.com/python-control/python-control/pull/617#discussion_r625361328

billtubbs, 2021-08-15:
Okay, thanks. So more of an accident than an intention.

billtubbs, 2021-08-15:
Okay, I've run into the same problem from last time I looked at this.  The [`sgrid()`](https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/grid.py#L68) function is not compatible with the plotting paradigm we wanted to employ for figures with multiple subplots:

```python
fig, axes = plt.subplots(2, 1)
zgrid(ax=axes[0, 0])  # this works
sgrid(ax=axes[1, 0])  # this won't
plt.show()
```

This won't work because `sgrid` has to produce a `PolarAxes.PolarTransform` which you have to do at the point of creating the axis, not after creating it:

https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/grid.py#L88-L93

Also, I don't think there's a way to remove an existing axis and then replace it with the desired polar axis (could be wrong on this point).

Welcome comments on this but I think there is one solution...

billtubbs, 2021-08-15:
The only solution I can think of right now is to require a slightly different set of arguments when adding an `sgrid` to an existing figure. Based on [this example script](https://matplotlib.org/stable/gallery/axisartist/demo_curvelinear_grid.html) from the documentation, I think this might work but I haven't tried implementing it yet:

```python
fig, axes = plt.subplots(2, 1)
zgrid(ax=axes[0, 0])  # this works
sgrid(fig=fig, position=(1, 0))  # this should be do-able
plt.show()
```

From the documentation, it sounds like the subplots method was intended as a convenience method for making plots of the same type only:

> This utility wrapper makes it convenient to create common layouts of subplots

so I think this means you can only create figures with subplots all of the same type.  It does have two keyword arguments for creating different types of axes and grids, [subplot_kw](https://matplotlib.org/devdocs/api/_as_gen/matplotlib.pyplot.subplots.html#matplotlib-pyplot-subplots) and [gridspec_kw](https://matplotlib.org/devdocs/api/_as_gen/matplotlib.pyplot.subplots.html#matplotlib-pyplot-subplots) but I'm not sure you can specify different types for each subplot (I will check this).

billtubbs, 2021-08-15:
Question.  I'm in the process of replacing `_sgrid_func` with `sgrid` and I noticed that `_sgrid_func` uses two other functions in `rlocus.py`:

https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/rlocus.py#L712-L718

https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/rlocus.py#L750-L754

Shall I delete these functions (they were only used by `_sgrid_func`) or should these features be carried over?

sawyerbfuller, 2021-08-16:
Hi bill, I’ve never seen them used anywhere else, and as private functions
I think you can delete them if they don’t show up in anywhere else in the
library.

bnavigator, 2021-08-16:
> ```python
> fig, axes = plt.subplots(2, 1)
> zgrid(ax=axes[0, 0])  # this works
> sgrid(ax=axes[1, 0])  # this won't
> plt.show()
> ```

> ```python
> fig = plt.figure()
> sgrid(fig=fig, position=(2, 1, 1))  # this should be do-able
> zgrid(fig=fig, position=(2, 1, 2))  # add the same functionality to zgrid
> plt.show()
> ```

You can always get the parent figure of a given axis by [`ax.get_figure()`](https://matplotlib.org/stable/api/_as_gen/matplotlib.artist.Artist.get_figure.html#matplotlib.artist.Artist.get_figure) and the position by [`ax.get_subplotspec()`](https://matplotlib.org/stable/api/_as_gen/matplotlib.axes.SubplotBase.html#matplotlib.axes.SubplotBase.get_subplotspec). So whatever you can achieve by the second form should also be possible with the first form.

bnavigator, 2021-08-16:
> This won't work because `sgrid` uses a `PolarAxes.PolarTransform` to create the axis, which you have to do at the point of creating the axis, not after one already exists:
> 
> https://github.com/python-control/python-control/blob/6b7a9c78f16ab04e009be7155308294080d8a5d1/control/grid.py#L88-L93
> 
> Also, I don't think there's a way to remove an existing axis and then replace it with the desired polar axis (could be wrong on this point).

The [MPL Curvilinear grid demo](https://matplotlib.org/stable/gallery/axisartist/demo_curvelinear_grid.html) where this was taken from has changed a little bit since then. Maybe this needs an update too.

billtubbs, 2021-08-16:
> You can always get the parent figure of a given axis by ax.get_figure() and the position by ax.get_subplotspec(). So whatever you can achieve by the second form should also be possible with the first form.

The problem is not getting access to the figure, it is that the axes has to be created in a certain way by `sgrid`.  `sgrid` cannot work with an axis that is already created so it has to be a blank figure (before the axis was created).

bnavigator, 2021-08-16:
I am not sure about this. Maybe you can shuffle around existing axes objects together with a newly created parasite axis using the right combination of [`delaxes(ax)`](https://matplotlib.org/stable/api/figure_api.html#matplotlib.figure.Figure.delaxes), [`fig.add_subplot(ax)`](https://matplotlib.org/stable/api/figure_api.html#matplotlib.figure.Figure.add_subplot), etc.

billtubbs, 2021-08-16:
Thanks, [`delaxes(ax)`](https://matplotlib.org/stable/api/figure_api.html#matplotlib.figure.Figure.delaxes) looks promising.  I will check out later this week.