python-control/python-control #635: real, imag, freq = nyquist_plot(sys)

iljastas, 2021-06-26:
Hi guys,

I get an error at nyquist_plot or nyquist:

`import control
from control import nyquist, nyquist_plot
import numpy as np
num = 5
den = [1,6,11,6]

# Creating a transfer function G = num/den
G = control.tf(num,den) 

try:
    real, imag, freq = nyquist(G, omega=np.linspace(0,10, 100), plot=True)
except:
    real, imag, freq = nyquist_plot(G, plot=True)`

`Traceback (most recent call last):
  File "/home/ilja/02_diss_ws/src/06_look_ahead_distance/test_nyquist.py", line 13, in <module>
    real, imag, freq = nyquist(G, omega=np.linspace(0,10, 100), plot=True)
TypeError: 'int' object is not iterable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ilja/02_diss_ws/src/06_look_ahead_distance/test_nyquist.py", line 15, in <module>
    real, imag, freq = nyquist_plot(G, plot=True)
TypeError: 'int' object is not iterable`

The return arguments are not correct anymore. It used to work, but not now.

I'm using:
control-0.9.0 kiwisolver-1.3.1 matplotlib-3.3.4 numpy-1.19.5 pillow-8.2.0 pyparsing-2.4.7 python-dateutil-2.8.1 scipy-1.5.4 six-1.16.0

Does anybody have an idea?

iljastas, 2021-06-26:
solved it by from control.matlab.wrappers import nyquist ... it used to be another path

bnavigator, 2021-06-26:
Hi @iljastas.

Glad you figured it out. Actually, the "correct" import path would be `from control.matlab import nyquist`

The TypeError is due to the different return signature of [control.matlab.nyquist()](https://python-control.readthedocs.io/en/0.9.0/generated/control.matlab.nyquist.html) vs [control.nyquist_plot()](https://python-control.readthedocs.io/en/0.9.0/generated/control.nyquist_plot.html)

closed 2021-06-26 by bnavigator