python-control/python-control #636: Example failing in Jupyter Lab on Linux Mint

mhassell, 2021-07-06:
I was working through the example in https://python-control.readthedocs.io/en/latest/optimal.html and I found that I am not able to get the results to work.  My resulting state just seems to keep drifting left from the target lane.  Am I missing something here?  Run on Linux Mint, Python 3, in Jupyter Lab.

Notebook is here:  https://github.com/mhassell/control_issue/blob/main/example.ipynb

Code/images: https://github.com/mhassell/control_issue/

As a sanity check I also ran it as a script, not_jupyter.py, using python3.  Same deal.  Saved the png as from_script_not_jupyter.png

bnavigator, 2021-07-07:
Hi, the optimal example in the docs might be a bit outdated. Did you try with [examples/steerining-optimal.py](https://github.com/python-control/python-control/blob/master/examples/steering-optimal.py) as well? The definition of the cost functions and optimization methods were adjusted a bit.

ThomasGiavasopoulos, 2022-02-02:
Thank you! Had the same issue with the outdated code. Thanks again!

closed 2022-03-30 by murrayrm