python-control/python-control #656: \n shall be \r\n\ in ipython for windows.

Barzi2001, 2021-09-26:
Perhaps `\n` shall be replaced with `\r\n` when running under `Windows`.
I tried to circumnavigate this problem by changing `c.InteractiveShell.separate_in` both `iPython profile_default.py` and `ipython_kernel_config` with `\r\n`,`\\n` and `\\r\\n` but it didn't solve the problem. 

You can easily reproduce the problem as it follows. 

```
In [1]: import control.matlab as ct
In [2]: s=ct.tf('s')
In [3]: %whos
Variable   Type                Data/Info
----------------------------------------
ct         module              <module 'control.matlab' <...>ol\\matlab\\__init__.py'>
s          TransferFunction    \ns\n-\n1\n
```
As you can see, in the way TransferFunction is displayed the `\n` should be corrected with `\r\n`.
I have `Python 3.9.7` with `iPython 7.28.0` on `Windows 10`.

bnavigator, 2021-09-28:
I cannot reproduce a problem on my Windows VM. Python interactive shells print newlines correctly without carriage return:

```
(controlenv) C:\Users\greiner>python
Python 3.9.7 | packaged by conda-forge | (default, Sep 23 2021, 07:24:41) [MSC v.1916 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> import control
>>> s = control.tf([1,0],[1])
>>> str(s)
'\ns\n-\n1\n'
>>> print(s)

s
-
1

>>>
```

```
(controlenv) C:\Users\greiner>ipython
Python 3.9.7 | packaged by conda-forge | (default, Sep 23 2021, 07:24:41) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 7.27.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import control

In [2]: s = control.tf([1,0],[1])

In [3]: %whos
Variable   Type                Data/Info
----------------------------------------
control    module              <module 'control' from 'C<...>s\\control\\__init__.py'>
s          TransferFunction    \ns\n-\n1\n

In [4]: print(s)

s
-
1

In [5]: s
Out[5]: TransferFunction(array([1, 0]), array([1]))
```

Barzi2001, 2021-09-29:
My bad.  `print(s)` works correctly on my computer as well.
The thing is that I erroneously thought that `%whos` would have listed transfer functions it in a "nice" format  (for "nice" I mean the same way it is displayed with the `print` method, without all those `\n`).  
I am a newbie in Python though :)

closed 2021-09-29 by Barzi2001