python-control/python-control #675: Test fails after installation via conda

alkhwarizmi, 2021-11-30:
Dear all,

just tried to install the library with the command `conda install control`.

As I was not able to import the control package, I run the test with the command: `pytest --pyargs control -v`

The test results are shown below, basically something fails.

Any suggestion on how I could solve this?

Sincerely,

```
================================================= test session starts =================================================
platform win32 -- Python 3.7.6, pytest-5.3.5, py-1.8.1, pluggy-0.13.1 -- C:\Anaconda3\python.exe
cachedir: .pytest_cache
hypothesis profile 'default' -> database=DirectoryBasedExampleDatabase('C:\\Users\\XXX\\.hypothesis\\examples')
rootdir: C:\Users\XXX
plugins: hypothesis-5.5.4, arraydiff-0.3, astropy-header-0.1.2, cov-2.12.1, doctestplus-0.5.0, filter-subpackage-0.1.1, mock-3.6.1, openfiles-0.4.0, remotedata-0.3.2
collected 2337 items / 1 error / 2336 selected

======================================================= ERRORS ========================================================
____________________________________ ERROR collecting tests/xferfcn_input_test.py _____________________________________
C:\Anaconda3\lib\site-packages\pluggy\hooks.py:286: in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
C:\Anaconda3\lib\site-packages\pluggy\manager.py:93: in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
C:\Anaconda3\lib\site-packages\pluggy\manager.py:87: in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
C:\Anaconda3\lib\site-packages\_pytest\python.py:231: in pytest_pycollect_makeitem
    res = list(collector._genfunctions(name, obj))
C:\Anaconda3\lib\site-packages\_pytest\python.py:395: in _genfunctions
    self.ihook.pytest_generate_tests.call_extra(methods, dict(metafunc=metafunc))
C:\Anaconda3\lib\site-packages\pluggy\hooks.py:324: in call_extra
    return self(**kwargs)
C:\Anaconda3\lib\site-packages\pluggy\hooks.py:286: in __call__
    return self._hookexec(self, self.get_hookimpls(), kwargs)
C:\Anaconda3\lib\site-packages\pluggy\manager.py:93: in _hookexec
    return self._inner_hookexec(hook, methods, kwargs)
C:\Anaconda3\lib\site-packages\pluggy\manager.py:87: in <lambda>
    firstresult=hook.spec.opts.get("firstresult") if hook.spec else False,
C:\Anaconda3\lib\site-packages\_pytest\python.py:126: in pytest_generate_tests
    metafunc.parametrize(*marker.args, **marker.kwargs)
C:\Anaconda3\lib\site-packages\_pytest\python.py:981: in parametrize
    ids = self._resolve_arg_ids(argnames, ids, parameters, item=self.definition)
C:\Anaconda3\lib\site-packages\_pytest\python.py:1035: in _resolve_arg_ids
    ids = idmaker(argnames, parameters, idfn, ids, self.config, item=item)
C:\Anaconda3\lib\site-packages\_pytest\python.py:1192: in idmaker
    for valindex, parameterset in enumerate(parametersets)
C:\Anaconda3\lib\site-packages\_pytest\python.py:1192: in <listcomp>
    for valindex, parameterset in enumerate(parametersets)
C:\Anaconda3\lib\site-packages\_pytest\python.py:1179: in _idvalset
    if ids is None or (idx >= len(ids) or ids[idx] is None):
E   TypeError: 'dict_keys' object is not subscriptable
================================================== warnings summary ===================================================
C:\Anaconda3\lib\site-packages\_pytest\mark\structures.py:327
  C:\Anaconda3\lib\site-packages\_pytest\mark\structures.py:327: PytestUnknownMarkWarning: Unknown pytest.mark.timeout - is this a typo?  You can register custom marks to avoid this warning - for details, see https://docs.pytest.org/en/latest/mark.html
    PytestUnknownMarkWarning,

-- Docs: https://docs.pytest.org/en/latest/warnings.html
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 1 error during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
============================================= 1 warning, 1 error in 8.32s =============================================
```

bnavigator, 2021-11-30:
Hi,

if updating pytest to version 6 is not an option for you, could you try to replace

https://github.com/python-control/python-control/blob/382a297868e91eaf3867874b3602be585cf68fbc/control/tests/xferfcn_input_test.py#L60

with

```python
@pytest.mark.parametrize("num, fun", cases.values(), ids=list(cases.keys()))
```

and see if that helps?

> As I was not able to import the control package,

The test suite collection only fails at a very late stage, so that does not seem to be related.

alkhwarizmi, 2021-11-30:
Hi,

regarding the test, I upgraded to version 6 (which upgraded many packages). Now (before I think test failed immediately) the tests have been run completely. Most of them complete successfully, some are skipped and some xfailed (?) below the report.

```
================================================== warnings summary ===================================================
..\..\Anaconda3\lib\site-packages\control\tests\rlocus_test.py:79
  C:\Anaconda3\lib\site-packages\control\tests\rlocus_test.py:79: PytestUnknownMarkWarning: Unknown pytest.mark.timeout - is this a typo?  You can register custom marks to avoid this warning - for details, see https://docs.pytest.org/en/stable/mark.html
    @pytest.mark.timeout(2)

tests/timeresp_test.py::TestTimeresp::test_step_info[siso_tf_type1-ltisys-yfinal]
  C:\Anaconda3\lib\site-packages\control\timeresp.py:885: RuntimeWarning: invalid value encountered in sign
    sgnInf = np.sign(InfValue.real)

-- Docs: https://docs.pytest.org/en/stable/warnings.html
========================= 2468 passed, 15 skipped, 6 xfailed, 2 warnings in 84.24s (0:01:24) ==========================
```

However, after a complete restart of the Python kernel, the statement  `import control` works and also the package I was trying to use seems to be working

Thank you very much for your help. Very much appreciated.

bnavigator, 2021-11-30:
> Most of them complete successfully, some are skipped and some xfailed (?) below the report.

Yes that's the expected behavior. The first warning would go away with pytest-timeout installed, but it is not relevant for you.

closed 2021-11-30 by bnavigator