python-control/python-control #687: python-control  0.9.1 fails in `test_nyquist_indent` on aarch64

ggardet, 2022-01-04:
python-control  0.9.1 fails in `test_nyquist_indent` on aarch64:
```
[  132s] =================================== FAILURES ===================================
[  132s] _____________________________ test_nyquist_indent ______________________________
[  132s] 
[  132s]     def test_nyquist_indent():
[  132s]         # FBS Figure 10.10
[  132s]         s = ct.tf('s')
[  132s]         sys = 3 * (s+6)**2 / (s * (s+1)**2)
[  132s]         # poles: [-1, -1, 0]
[  132s]     
[  132s]         plt.figure();
[  132s]         count = ct.nyquist_plot(sys)
[  132s]         plt.title("Pole at origin; indent_radius=default")
[  132s]         assert _Z(sys) == count + _P(sys)
[  132s]     
[  132s]         # first value of default omega vector was 0.1, replaced by 0. for contour
[  132s]         # indent_radius is larger than 0.1 -> no extra quater circle around origin
[  132s]         count, contour = ct.nyquist_plot(sys, plot=False, indent_radius=.1007,
[  132s]                                          return_contour=True)
[  132s]         np.testing.assert_allclose(contour[0], .1007+0.j)
[  132s]         # second value of omega_vector is larger than indent_radius: not indented
[  132s] >       assert np.all(contour.real[2:] == 0.)
[  132s] E       AssertionError: assert False
[  132s] E        +  where False = <function all at 0xffff92ec1d30>(array([0.1001..., 0.        ]) == 0.0
[  132s] E        +    where <function all at 0xffff92ec1d30> = np.all
[  132s] E           +array([0.1001835 , 0.10017386, 0.10016404, 0.10015404, 0.10014385,\n
[  132s] E           +       0.10013347, 0.1001229 , 0.10011213, 0.10010116, 0.10008998,\n
[  132s] E           +       0.10007859, 0.10006699, 0.10005517, 0.10004313, 0.10003086,\n
[  132s] E           +       0.10001837, 0.10000564, 0.09999267, 0.09997946, 0.099966  ,\n
[  132s] E           +       0.09995229, 0.09993832, 0.09992409, 0.0999096 , 0.09989483,\n
[  132s] E           +       0.09987978, 0.09986445, 0.09984884, 0.09983293, 0.09981672,\n
[  132s] E           +       0.09980021, 0.09978339, 0.09976625, 0.09974879, 0.099731  ,\n...
[  132s] E           
[  132s] E           ...Full output truncated (195 lines hidden), use '-vv' to show)
[  132s] 
[  132s] control/tests/nyquist_test.py:202: AssertionError
```

bnavigator, 2022-01-04:
I think this is similar to https://github.com/python-control/Slycot/issues/156: There is a BLAS/LAPACK problem of aarch64 in openSUSE.

bnavigator, 2022-01-04:
I was wrong. This test fails because it assumes that the default frequency vector calculation uses the poles at `-1` to determine the logspace to start at `0.1`. But some architectures return the floating point values of the poles as *slightly* closer to `0` and then the logspace starts one decade lower.

PR #688 (2022-01-05): Round to nearest integer decade for default omega vector
Fixes #687.

The docstring for the `_default_frequency_range()` function says

> This code looks at the poles and zeros of all of the systems that
> we are plotting and sets the frequency range to be one decade above
> and below the min and max feature frequencies, rounded to the nearest
> integer.

The old code rounded to the floor and ceiling of the range, which occasionally adds another decade to one or the other end of the frequency range if the feature falls *just* inside of the outermost decade.

The new code rounds to the nearest  [... cut]
closed 2022-01-05 by bnavigator