python-control/python-control #690: Create matlab's equivalent to bandwidth.

Kreijstal, 2022-01-09:
Is it possible to create matlab's equivalent to bandwidth in control?
https://www.mathworks.com/help/control/ref/lti.bandwidth.html
or are there any alternatives available?

murrayrm, 2022-01-09:
Definitely possible, although this can also be done with the output from `bode_plot` and the Numpy `argmin` function, I suspect.

If you (or someone) want to contribute code to do this, we can put it either in the main package or in the MATLAB compatibility module.

Kreijstal, 2022-01-10:
Here is _roughly_ how I solved it for me, I don't have time for providing a pull request,, but maybe someone writing the pull request can get inspired. 

```python
import scipy
import numpy as np
def bandwidth(sys,dbgain=-3):
  result=scipy.optimize.root(lambda s:control.matlab.freqresp(sys,s)[0][0]-control.matlab.dcgain(sys)*10**(dbgain/20),1)
  if result.success:
    return np.abs(result.x)[0]
  else:
    raise Exception(result.message)
```

roryyorke, 2024-07-14:
Looks like merged #889 closes this?

Kreijstal, 2024-07-14:
I have not used control in years, but I am happy it got solved

closed 2024-07-14 by Kreijstal