python-control/python-control #691: Discontinuous Nyquist Plot

JayEy, 2022-01-15:
Today I used the following code to produce the nyquist plot of an arbitrary transfer function:

```
import numpy as np
from control.matlab import *
import matplotlib.pyplot as plt

a0 = 1.; a1 = 10.; a2 = 31.; a3 = 30
b0 = 1.
num = np.array([b0])
den = np.array([a3, a2, a1, a0])
system_tf = tf(num,den)
ctrl_tf = tf([5, 1],[5, 0])
G0 = ctrl_tf*system_tf
Gw = feedback(G0,1)
nyquist(Gw)
plt.show()

```
and I got this strange discontinuous Nyquist plot:
[image]

the bode plot looks correct and I found that both plots use different functions to evaluate the frequency response. So I think there might be a bug in the implementation of the nyquist plot.

Here is the expected Nyquist plot that I could obtain from Matlab.
[image]

bnavigator, 2022-01-15:
Hi @JayEy, thanks for flagging this.

You can work around the ugly resolution of the nyquist contour by decreasing the value for the [indent_radius](https://python-control.readthedocs.io/en/0.9.1/generated/control.nyquist_plot.html) parameter:

```
control.nyquist_plot(Gw, indent_radius=1e-2)
```

[image]

JayEy, 2022-01-15:
Hi @bnavigator , thanks! Looks just fine with the reduced indent radius.

murrayrm, 2022-01-15:
It would be good to think through the algorithm we are using for the indent radius (currently just setting it to a fixed value) to see if we can avoid situations like this.  I'm flagging this issue as an enhancement in case someone has time to take a look at this.

We might also look at this example as part of the discussion in #671.

sawyerbfuller, 2022-01-18:
@JayEy maybe one issue is that you are passing the closed-loop system to the nyquist plot, but its use is to assess stability given only the open-loop system. The open-loop plot `nyquist(G0)` appears to give a more reasonable plot (see below). 

@murrayrm One idea would be to calculate the indent radius on a per-pole basis. By partial fraction expansion, the transfer function can be approximated near a pole p by G(s) = A/(s-p)**k + B, where k is the order of the pole. One could evaluate the transfer function at two nearby points to estimate A and B. Then, to have the contour evaluate to a radius of approximately R, then the indent radius r should satisfy A/(r-p)**k + B = R, or r = p + (A/(R-B))**-k. 

[image]

sawyerbfuller, 2022-01-18:
Hi @JayEy these look the basically the same to me. Indeed one way to think abotu the nyquist plot is as the path of the complex frequency response (i.e. the bode plot), but with mag and phase plotted against each other, instead of against the frequency. 

Unlike Matlab, python-control takes the additional step of depicting the path of the nyquist contour as the frequency response passes near poles that are near the imaginary axis, which we think helps in understanding the path of the complete contour. Those corners are the result of that. really what is happening is that the contour extends off to infinity, and that large, backwards "C" shape is intended to represent that.

LukasMueller187, 2022-02-17:
Hi there,
I ran into the same "problem" today, thanks for clarification @sawyerbfuller. I think the contour that is intended to show what is happening as the frequency passes near the poles should be depicted otherwise. I'd suggest decreasing the line width, color to gray and maybe line-style to dotted.

I was heavily confused by the following plot (it circles twice):
[image]

This is the nyquist plot without the frequency response near poles:
[image]

Regards

murrayrm, 2022-04-16:
The original issue identified in this post seems to be fixed in PR #722.  The curve with default parameters now looks like this:
[image]

Note that portions of the curve corresponding to indented regions now plotted in a different style.

closed 2022-04-16 by murrayrm