python-control/python-control #694: Optimal control results should include the final cost achieved

murrayrm, 2022-01-26:
The the optimal control module (`ct.optimal`), it would be useful if the functions that compute an optimal trajectory returned the cost associated with the final trajectory that is returned, so that it doesn't have to be recomputed.  (Perhaps also a function to compute the cost along a curve according to an arbitrary cost function.)

ThomasGiavasopoulos, 2022-03-02:
Is there currently an option to return the value of the cost associated to a final trajectory given a time horizon?

murrayrm, 2022-03-03:
The return object from `ct.optimal.solve_ocp` is a [OptimizeResult](https://docs.scipy.org/doc/scipy/reference/generated/scipy.optimize.OptimizeResult.html#scipy.optimize.OptimizeResult) object, so `result.fun` will get you the value of the function.  Whoever handles this enhancement should change that to be something more meaningful (eg, `result.cost` as a property that maps to `result.fun`).

murrayrm, 2022-03-18:
FYI, I have added this to an upcoming PR.

ThomasGiavasopoulos, 2022-03-19:
> FYI, I have added this to an upcoming PR.

That sounds really useful. Hope that it gets added soon!

closed 2022-03-30 by murrayrm