python-control/python-control #700: Bug with forced_response() for MIMO systems

Siarzis, 2022-02-01:
I have been trying to use the forced_response() function for a MIMO system with 2 inputs and 2 outputs defined by its state space representation. However, when I passed the U=[U1, U2] input vector to the forced_response() function, it returned only one output which was also incorrect. I created a custom forced_response_mimo() function which was based on the step_response() function that supports MIMO systems defined by its state space representation and it performed as expected. Is it possible to test and resolve this issue?

bnavigator, 2022-02-01:
Hi, can you please give some example code?

Siarzis, 2022-02-02:
Let us consider the following 2x1 MIMO system, defined by the following arrays:
```
A = [[-0.5572, -0.7814],
     [0.7814, 0]
    ]

B = [[1, -1],
     [0, 2]
    ]

C = [[1.9691, 6.4493]
    ]

D = 0
```

Then we create the state space model:

```
test_system = ss(A, B, C, D)
```

The following code calculates and plots the step response of the previous 2x1 MIMO system:
```
response = step_response(test_system)

fig, (ax1, ax2) = plt.subplots(1, 2, sharey=True)

ax1.plot(response.time, response.outputs[0][0])
ax1.set_title('Input 1')

ax2.plot(response.time, response.outputs[0][1])
ax2.set_title('Input 2')

plt.show()
```

The previous code produces the following plot which is the expected step response of each input for the particular system:

[image]

Then, we can use following code to investigate the functionality of the `forced_response()`:

```
T = np.arange(0, 30, 0.1, dtype=float).tolist()
U = [np.ones_like(T).tolist(), np.ones_like(T).tolist()]

t, y = forced_response(test_system, T=T, U=U)

fig, (ax1, ax2) = plt.subplots(1, 2, sharey=True)

ax1.plot(t, y[0])
ax1.set_title('Input 1')

ax2.plot(t, y[1])
ax2.set_title('Input 2')

plt.show()
```

As you see, we have provided the input array `U` with size (2, 300) which is filled with ones. In this way, we can evaluate if the `forced_response()` has the same output with the `step_response()` since we have provided the `forced_response()` with 2 input arrays of ones, as `step_response()` does. However, if we run the above code we get the following error:

> File "xxx", line 235, in \<module\> ax2.plot(t, y[1])
> IndexError: index 1 is out of bounds for axis 0 with size 1

This means that the forced response has produced only one output. We can verify it by printing the size of the `y` numpy array which is (1, 300). Plotting only the `y[0]` output produces the following figure:

[image]

Of course, this is not the expected step response. Probably, it should match at least on of the figures that was produced with the `step_response()` but it does not.

I resolved this issue by myself by editing the provided `step_response()` of the control library and adjusting it to accept any input `U` and not an array of ones like the `step_response()` does. Since we have verified that the `step_response()` is created is such a way that supports multiple inputs it is a reliable template for this task.

bnavigator, 2022-02-02:
```
T = np.arange(0, 30, 0.1, dtype=float).tolist()
U = [np.ones_like(T).tolist(), np.ones_like(T).tolist()]

t, y = forced_response(test_system, T=T, U=U)
```

But that is not the equivalent of the step response for individual inputs. It is the response to stepping both inputs simultaneously.

The equivalent to `step_response()` would be

```
T = np.arange(0, 30, 0.1, dtype=float).tolist()
U1 = [np.ones_like(T).tolist(), np.zeros_like(T).tolist()]
U2 = [np.zeros_like(T).tolist(), np.ones_like(T).tolist()]

t, y1 = forced_response(test_system, T=T, U=U1)
t, y2 = forced_response(test_system, T=T, U=U2)
y = np.vstack([y1,y2])
```
which produces the expected output.

----------

> File "xxx", line 235, in <module> ax2.plot(t, y[1])
> IndexError: index 1 is out of bounds for axis 0 with size 1

The system only has one output. `y = Cx+D` has only one row.

closed 2022-02-02 by bnavigator