python-control/python-control #715: Incorrect rootlocus plot

warlock31415, 2022-03-27:
I'm using the control package in a modified jupyter/scipy docker container. The modification to the container were just to install the controls package (`RUN pip3 install control`). When i try to plot the root locus of a inverted pendulum with 2 roots at 0, it outputs the wrong rootlocus.

### Minimal working example:
```
import numpy as np
import control

A = np.array([[0,1,0,0],[0,0,-1,0],[0,0,0,1],[0,0,5,0]])
B = np.array([[0],[1],[0],[-2]])
C = np.array([1,0,0,0])
D = 0
sys = control.ss(A,B,C,D)
r = control.root_locus(sys, xlim=[-2,2],ylim=[-7,7])
```
The above code results in the following plot
[image]
When the same system in plotted in octave, it gives the following rlocus
[image]

As can be seen there should be no branch immediately to the left of 0

murrayrm, 2022-03-27:
What version of python-control are you running?  (Look at `control.__version__`.)  In 0.9.1 I get the following (dropping `xlim` and `ylim`):

[image]

warlock31415, 2022-03-27:
Thats's interesting i'm running 0.9.1 as well:
[image]

I tried using it natively with the same code and got the same incorrect plot. I had to use the axes limits because the plot scales bad. Just to make sure, here's the code I'm using:
```
#! /usr/bin/python3

import numpy as np
import control
import matplotlib.pyplot as plt
A = np.array([[0,1,0,0],[0,0,-1,0],[0,0,0,1],[0,0,5,0]])
B = np.array([[0],[1],[0],[-2]])
C = np.array([1,0,0,0])
D = 0
print(control.__version__)
sys = control.ss(A,B,C,D)
control.root_locus(sys)
plt.show()
```
[image]

And zoomed in looks like the image i linked earlier.

murrayrm, 2022-03-27:
I was able to reproduce the error above using [Google colab]( https://colab.research.google.com/drive/1e45u4oKRyR0XSA6luhJEjx5PKsjDrg-4?usp=sharing).

I'll have to look into this, but could have something to do with the version of NumPy or possibly lack of slycot?  Still, very strange.

murrayrm, 2022-03-27:
Reproduced on my local machine using python=3.7 numpy=1.21 matplotlib=3.2 control=0.9.1.

warlock31415, 2022-03-27:
Thanks for getting on this!

warlock31415, 2022-03-28:
Added slycot to my docker image. Now getting correct root locus. So might be a slycot issue like you said earlier.

murrayrm, 2022-03-28:
I chased this down a bit more.  The difference seems to come from the way that slycot creates the transfer function representation of the system versus the way that SciPy carries out the calculation (via `scipy.signal.ss2tf`).

Slycot:
* Poles: array([ 2.23606798e+00, -2.23606798e+00,  1.26650220e-17,  1.26650216e-17])
* Zeros: array([ 1.73205081, -1.73205081])

SciPy:
* Poles: array([-2.23606798,  2.23606798,  0.        ,  0.        ])
* Zeros:  array([-2.68871620e+14,  1.73205081e+00, -1.73205081e+00])

Interestingly, SciPy throws a warning when computing the poles:
```
/Users/murray/anaconda3/envs/python3.9-slycot/lib/python3.9/site-
packages/scipy/signal/filter_design.py:1630: BadCoefficients: Badly
conditioned filter coefficients (numerator): the results may be meaningless
```
 
So it looks like the problem is the (fast, stable) zero at -2.689e+14, which completely throws off the root locus plot.  Not sure how to fix without implementing our own transfer function calculation rather than rely on SciPy.  

I looked at the source code for the SciPy function and the basic calculation is:
```
den = poly(A)
num[k] = poly(A - dot(B, C[k:k+1, :])) + (D[k] - 1) * den
```

The Slycot computation for `tb04ad` (the function that gets calls) says:

> The method for transfer matrices factorized by rows will be
  described here: T(s) factorized by columns is dealt with by
  operating on the dual of the original system.  Each row of
  T(s) is simply a single-output relatively left prime polynomial
  matrix representation, so can be calculated by applying a
  simplified version of the Orthogonal Structure Theorem to a
  minimal state-space representation for the corresponding row of
  the given system. A minimal state-space representation is obtained
  using the Orthogonal Canonical Form to first separate out a
  completely controllable one for the overall system and then, for
  each row in turn, applying it again to the resulting dual SIMO
  (single-input multi-output) system. Note that the elements of the
  transformed matrix A so calculated are individually scaled in a
  way which guarantees a monic denominator polynomial.

murrayrm, 2023-03-27:
Marking this as a bug/issue in scipy.signal.ss2tf => nothing to be done here.

closed 2023-03-27 by murrayrm