python-control/python-control #716: SyntaxError: invalid syntax in python2.7.17

GuancongLuo, 2022-03-29:
Python 2.7.17 (default, Apr 15 2020, 17:20:14)
[GCC 7.5.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import control
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/bot/.local/lib/python2.7/site-packages/control/__init__.py", line 49, in <module>
    from .bdalg import *
  File "/home/bot/.local/lib/python2.7/site-packages/control/bdalg.py", line 57, in <module>
    from . import xferfcn as tf
  File "/home/bot/.local/lib/python2.7/site-packages/control/xferfcn.py", line 62, in <module>
    from .lti import LTI, common_timebase, isdtime, _process_frequency_response
  File "/home/bot/.local/lib/python2.7/site-packages/control/lti.py", line 18, in <module>
    from . import config
  File "/home/bot/.local/lib/python2.7/site-packages/control/config.py", line 51
    if self.__contains__(f"deprecated.{key}"):
                                           ^
SyntaxError: invalid syntax
>>>

bnavigator, 2022-03-29:
Python 2.7 is end-of-life since January 2020 and not supported by python-control since version 0.9.1

bnavigator, 2022-03-29:
Please use a more recent Python. It looks like you installed python-control into your Python 2.7 user site-lib with `pip install control` using the system wide `pip`. Make sure `pip` points to a Python 3 version or use `pip3`. Use `python3` instead of `python` afterwards. (See also [PEP 294](https://peps.python.org/pep-0394/))

closed 2022-03-29 by bnavigator