python-control/python-control #725: control.matlab.series is inconsistent with MATLAB

sylvain-bougnoux, 2022-04-20:
The final state vector is composed differently in matlab vs python-control, eg for 2 state space systems, `ss1` & `ss2`, `ss3=series(sys2,sys1)`, aka `sys3=sys1*sys2` produces as state vector `x3=(x1 x2)` in matlab and `x3=(x2 x1)` in python-control.

murrayrm, 2022-04-21:
The block diagram for ss3 = series(sys1, sys2) is

   u --> [sys1] --> [sys2] --> y

For this, I would expect the states of the interconnected system to be the states of sys1 followed by the states of sys2, which is what you get.  Of course, if you use overloaded operators then you would write `sys3 = sys2 * sys1`.

The current ordering is also consistent with other operators:

* sys3 = series(sys1, sys2) gives sys3.x = [sys1.x, sys2.x]
* sys3 = parallel(sys1, sys2) gives sys3.x = [sys1.x, sys2.x]
* sys3 = feedback(sys1, sys2) gives sys3.x = [sys1.x, sys2.x]

It's easy enough to switch, but the current ordering seems consistent.  Thoughts from others?

sylvain-bougnoux, 2022-04-21:
You are right, I guess it is consistent & meaningful. The point is that it is different from matlab. Therefore maybe `control.series` should be what you have, and `control.matlab.series` might do the switch? (While `*` keeps `control.series`).

It is just in case porting a code from matlab while manipulating indexes is currently error prone.

murrayrm, 2023-03-27:
I looked into this further and it does not appear that the order of states in a series interconnection is documented in MATLAB.  Furthermore, making this be different for the `matlab` module would require essentially rewriting all of the block diagram algebra code in the `matlab` module.  Given the (hopefully) relatively rare cases where state order matters, I am going to leave the code unchanged.

closed 2023-03-27 by murrayrm