python-control/python-control #726: Any plan for implementing disk margins?

sylvain-bougnoux, 2022-04-20:
This technique is much more robust than gain & phase margins.

https://arxiv.org/pdf/2003.04771.pdf

https://www.youtube.com/watch?v=XazdN6eZF80&list=PLn8PRpmsu08qFLMfgTEzR8DxOPE7fBiin&index=2&ab_channel=MATLAB

murrayrm, 2022-04-21:
I'm not aware of any plans along these lines, but it would be great if you wanted to submit some code implementing this functionality.  Information on how to contribute is [here](https://github.com/python-control/python-control/wiki/How-to-contribute-with-a-pull-request).

sylvain-bougnoux, 2022-04-22:
I wish I could do it, currently I am not familiar enough with the algorithm. FTTB, one can find an implementation here (in julia)

https://github.com/JuliaControl/RobustAndOptimalControl.jl/blob/master/src/diskmargin.jl

roryyorke, 2022-04-24:
We do compute what this paper calls the S-based disk margin in [`stability_margins`](https://python-control.readthedocs.io/en/0.9.1/generated/control.stability_margins.html#control-stability-margins); it's called "Stability margin" in our docs.

Slycot already has a function to compute the transfer function matrix L-infinity norm (which is the H-infinity norm for stable systems), so the hard part to implement disk-margins is done.

roryyorke, 2022-04-27:
See also [the published form of the disk margin paper](https://www.researchgate.net/publication/349442739_An_Introduction_To_Disk_Margins), and [Matlab's diskmargin command](https://www.mathworks.com/help/robust/ref/diskmargin.html).  One of the paper's authors, the late Andrew Packard, was an author of Matlab's Robust Control Toolbox.

The SISO disk margins are straightforward, and I've got prototype code to find the disk margin, and the associated destabilising complex perturbation and destabilising LTI system.

The _MIMO_ version of disk margins needs structured singular values, or at least I'm pretty sure it does.  [SLICOT has AB13MD](http://slicot.org/objects/software/shared/doc/AB13MD.html), which computes an upper bound on structure singular values, and if we want MIMO disk margins, I guess we'd start there.  For now, one can get conservative upper bounds on (two of) these using `linfnorm` on the sensitivity and complementary sensitivity transfer functions.