python-control/python-control #727: Need helps to figure out the Error and how to fix it

shiling2007, 2022-04-24:
I am running the example from https://github.com/python-control/python-control/blob/master/examples/kincar-flatsys.py. And I am getting an error message as below,

---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
[<ipython-input-50-3158d2b11897>](https://localhost:8080/#) in <module>()
    130 # Simulation the open system dynamics with the full input
    131 t, y, x = ct.input_output_response(
--> 132     vehicle_flat, T, ud, x0, return_x=True)
    133 
    134 # Plot the open loop system dynamics

2 frames
[/usr/local/lib/python3.7/dist-packages/control/timeresp.py](https://localhost:8080/#) in _process_labels(labels, signal, length)
    681     # Make sure the signal list is the right length and type
    682     if len(labels) != length:
--> 683         raise ValueError("List of %s labels is the wrong length" % signal)
    684     elif not all([isinstance(label, str) for label in labels]):
    685         raise ValueError("List of %s labels must all  be strings" % signal)

ValueError: List of output labels is the wrong length**

The program is running in google Colab environment with python version 3.7.
Your help would be much appreciated.

roryyorke, 2022-04-24:
That example was updated recently.  Assuming you're running python-control version 0.9.1 (check with `import control as ct; print(ct.__version__)`, you need [the example file from 0.9.1](https://github.com/python-control/python-control/blob/0.9.1/examples/kincar-flatsys.py).

murrayrm, 2022-04-24:
Alternatively, you can edit the latest version of the example to be compatible with the earlier version of the code.  Here's the diff to implement:
```
--- a/examples/kincar-flatsys.py
+++ b/examples/kincar-flatsys.py
@@ -109,7 +109,7 @@ def plot_results(t, x, ud):
 # Create differentially flat input/output system
 vehicle_flat = fs.FlatSystem(
     vehicle_flat_forward, vehicle_flat_reverse, vehicle_update,
-    inputs=('v', 'delta'), outputs=('x', 'y', 'theta'),
+    inputs=('v', 'delta'), outputs=('x', 'y'),
     states=('x', 'y', 'theta'))
```
The change is that in the latest master the default for a flat system is that the default outputs are the flat outputs (in v0.9.1, the default outputs were the entire state, which didn't really make sense).  This was changed in PR #709.

closed 2022-04-24 by murrayrm

shiling2007, 2022-04-24:
I am good now. Thanks a lot.