python-control/python-control #730: MKL FATAL ERROR on Google Colab with mkl==2019.0

tvidano, 2022-05-01:
When using `control` version `0.9.1` on Google Colab and `mkl` version is `2019.0` the session crashes when running `ts2ss()` with the following runtime log:
[image]
Steps to recreate:
`!pip install control`
`!pip install slycot` (likely not related)
`!pip show mkl` (verify mkl version, my sessions have all been `2019.0` so far)
```
import control as ct
ct.tf2ss(ct.tf([1],[1,1]))
```
Session crashes with previous error in runtime logs.

The fix I have found is to update the `mkl` version to `2021.4.0`. I am not familiar enough with the version changes nor have I tried other `mkl` versions, so I cannot say what other versions work. 
`!pip install mkl==2021.4.0`

Probably should add a requirement on `mkl` version when installing `control` in Colab. 

I have not tried to reproduce this on a local Jupyter notebook.

bnavigator, 2022-05-02:
Hi,

python-control does not call any BLAS/LAPACK library routine directly. We don't have and won't add any requirement on MKL, OpenBLAS or any other implemetation of BLAS/LAPACK directly to python-control.

This is most likely an issue with the MKL/NumPy/Slycot combination of your instance in Google Colab. And as you have found out, it seems to be fixed by updating to a more recent version. I am not familiar with the innards of Colab, but it seems that `!pip install slycot` *is* the culprit here. On a regular Python environment this would download the source and compile the Fortran routines with the headers present on the system.

closed 2022-05-02 by bnavigator

bnavigator, 2022-05-02:
BTW, numpy on Colab says, that is not using MKL at all:

```python
import numpy as np
np.show_config()``
```

```
blas_mkl_info:
  NOT AVAILABLE
blis_info:
  NOT AVAILABLE
openblas_info:
    libraries = ['openblas', 'openblas']
    library_dirs = ['/usr/local/lib']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
    runtime_library_dirs = ['/usr/local/lib']
blas_opt_info:
    libraries = ['openblas', 'openblas']
    library_dirs = ['/usr/local/lib']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
    runtime_library_dirs = ['/usr/local/lib']
lapack_mkl_info:
  NOT AVAILABLE
openblas_lapack_info:
    libraries = ['openblas', 'openblas']
    library_dirs = ['/usr/local/lib']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
    runtime_library_dirs = ['/usr/local/lib']
lapack_opt_info:
    libraries = ['openblas', 'openblas']
    library_dirs = ['/usr/local/lib']
    language = c
    define_macros = [('HAVE_CBLAS', None)]
    runtime_library_dirs = ['/usr/local/lib']
Supported SIMD extensions in this NumPy install:
    baseline = SSE,SSE2,SSE3
    found = SSSE3,SSE41,POPCNT,SSE42,AVX,F16C,FMA3,AVX2
    not found = AVX512F,AVX512CD,AVX512_KNL,AVX512_KNM,AVX512_SKX,AVX512_CLX,AVX512_CNL,AVX512_ICL
```

If you compile Slycot with `-DBLA_VENDOR=Generic`, then you won't have to update MKL:

```python
%env SKBUILD_CONFIGURE_OPTIONS="-DBLA_VENDOR=Generic"
!pip install slycot control
import control as ct
```

```
env: SKBUILD_CONFIGURE_OPTIONS="-DBLA_VENDOR=Generic"
Collecting slycot
  Downloading slycot-0.4.0.0.tar.gz (1.5 MB)
     |████████████████████████████████| 1.5 MB 8.8 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting control
  Downloading control-0.9.1.tar.gz (357 kB)
     |████████████████████████████████| 357 kB 39.9 MB/s 
Requirement already satisfied: numpy in /usr/local/lib/python3.7/dist-packages (from slycot) (1.21.6)
Requirement already satisfied: scipy in /usr/local/lib/python3.7/dist-packages (from control) (1.4.1)
Requirement already satisfied: matplotlib in /usr/local/lib/python3.7/dist-packages (from control) (3.2.2)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /usr/local/lib/python3.7/dist-packages (from matplotlib->control) (3.0.8)
Requirement already satisfied: cycler>=0.10 in /usr/local/lib/python3.7/dist-packages (from matplotlib->control) (0.11.0)
Requirement already satisfied: python-dateutil>=2.1 in /usr/local/lib/python3.7/dist-packages (from matplotlib->control) (2.8.2)
Requirement already satisfied: kiwisolver>=1.0.1 in /usr/local/lib/python3.7/dist-packages (from matplotlib->control) (1.4.2)
Requirement already satisfied: typing-extensions in /usr/local/lib/python3.7/dist-packages (from kiwisolver>=1.0.1->matplotlib->control) (4.2.0)
Requirement already satisfied: six>=1.5 in /usr/local/lib/python3.7/dist-packages (from python-dateutil>=2.1->matplotlib->control) (1.15.0)
Building wheels for collected packages: slycot, control
  Building wheel for slycot (PEP 517) ... done
  Created wheel for slycot: filename=slycot-0.4.0-cp37-cp37m-linux_x86_64.whl size=1418275 sha256=99f50c169426a9233c293bd29c813c51e4d9abb480661c45964ae2885fe47a2b
  Stored in directory: /root/.cache/pip/wheels/fa/b2/19/1e6568896fbabaf2a03d2fb6575211d281afcb9e00ce026cf6
  Building wheel for control (setup.py) ... done
  Created wheel for control: filename=control-0.9.1-py2.py3-none-any.whl size=364703 sha256=294faf3306345fc5a496a380203c8378d79344075a7b0ae67358db5f613cbea0
  Stored in directory: /root/.cache/pip/wheels/b1/50/22/8005107713828997d1af664fbd629c7f935a1b87fc2f4ba6c1
Successfully built slycot control
Installing collected packages: slycot, control
Successfully installed control-0.9.1 slycot-0.4.0
```

```python
ct.tf2ss(ct.tf([1],[1,1]))
```

```
\[
\left(\begin{array}{rll|rll}
-1\phantom{.}&\hspace{-1em}&\hspace{-1em}\phantom{\cdot}&1\phantom{.}&\hspace{-1em}&\hspace{-1em}\phantom{\cdot}\\
\hline
1\phantom{.}&\hspace{-1em}&\hspace{-1em}\phantom{\cdot}&0\phantom{.}&\hspace{-1em}&\hspace{-1em}\phantom{\cdot}\\
\end{array}\right)
\]
```

Now you have to figure out how to get Colab to render LaTeX output like a real Jupyter Notebook or ignore it and use the regular `__repr__()`

Spogis, 2024-02-01:
Hi @bnavigator / Ben!

I saw your comments here and decided to get in touch. I am also trying to run a very simple code on Google Colab using the Control + slycot library, but the Kernel restarts (crashes).

Were you able to solve the problem?

Best regards,

Nicolas Spogis

__________
Colab Errors:

WARNING:root:kernel f5072b3f-42c0-4442-bb37-dd67744b63be restarted
KernelRestarter: restarting kernel (1/5), keep random ports
Intel MKL FATAL ERROR: Cannot load libmkl_avx2.so or libmkl_def.so.
INTEL MKL ERROR: /lib/x86_64-linux-gnu/libmkl_avx2.so: undefined symbol: mkl_sparse_optimize_bsr_trsm_i8.

__________
Code on Google Colab:
!apt-get install gfortran cmake --fix-missing
!apt-get install libblas-dev liblapack-dev

import os
os.environ['BLA_VENDOR']="Generic"
!echo Using $$BLA_VENDOR BLAS implementation

!pip install -v slycot

!pip install control

import control
import slycot
import matplotlib.pyplot as plt

# Define transfer function of the system
s = control.tf('s')

K1 = 5.0
T1 = 5.0
P1 = K1/(T1*s+1)

K2 = 10.0
T2 = 20.0
P2 = K2/(T2*s+1)

K3 = 5.0
T3 = 5.0
P3 = K3/(T3*s+1)

K4 = 5.0
T4 = 5.0
P4 = K4/(T4*s+1)

G = control.append(P1, P2, P3, P4) #Google Colab Crash HERE!

bnavigator, 2024-02-01:
Check https://github.com/python-control/Slycot/issues/183#issuecomment-1325030920

Spogis, 2024-02-01:
Now not crash, but i have this error:
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
[<ipython-input-11-beb8f34beafd>](https://localhost:8080/#) in <cell line: 1>()
----> 1 G = control.append(P1, P2, P3, P4)

2 frames
[/usr/local/lib/python3.10/dist-packages/slycot/transform.py](https://localhost:8080/#) in td04ad(rowcol, m, p, index, dcoeff, ucoeff, tol, ldwork)
    622                                            *(dcoeff.shape + expectedshape)),
    623                                        -5)
--> 624         out = _wrapper.td04ad_c(m,p,index,dcoeff,ucoeff,n,tol,ldwork)
    625     else:
    626         raise SlycotParameterError("Parameter rowcol had an illegal value", -1)

AttributeError: module 'slycot._wrapper' has no attribute 'td04ad_c'

This is my Google Colab: https://colab.research.google.com/drive/1hb9XDIVeMirQ2gevnaU282yFTs4-KS_t?usp=sharing

bnavigator, 2024-02-02:
I'm already getting a numpy error while importing control when running your colab.

bnavigator, 2024-02-02:
My own colab notebook still works, including your code:

https://colab.research.google.com/drive/1kKaK3jI_omOTrLCLQGzSs1uTowN5Cx_v?usp=sharing

Spogis, 2024-02-02:
Hi @bnavigator / Ben!

Thank you very very very much! 

You help me a lot.

As "The Cure" said! It's Friday, I'm in love!

Best regards,

Nicolas