python-control/python-control #749: Include a zpk2tf function

ovega14, 2022-07-07:
As it stands, a function that takes zeroes, poles, gain into numerator, denominator transfer function coefficient arrays only lives in `scipy.signal` or can be imported from `control.matlab`, and used to create a transfer function `my_tf` as follows:
`my_tf= tf(*zpk2tf(...))`
Why not instead can `python-control` have a method that directly takes zeroes, poles, gain into a transfer function, i.e.
`my_tf=zpk2tf(..)`?

sawyerbfuller, 2022-07-16:
Hi @ovega14 I agree it would be very nice to have this functionality outside of the matlab module. 

Its not clear to me how we could implement that functionality inside `TransferFunction` because the three-argument case is already taken by `TransferFunction(num, den, dt)`. Might be possible to include a keyword to make it explicit like `TransferFunction(zeros, poles, gain, [dt,] usezpk=True)`, but implementing that logic seems like it would be awkward. 

Probably most expedient is to just move the import of scipy's `zpk2tf' (and might as well bring in `zpk2ss` while you're at it) into the `LTI` module rather than the matlab module, and then they will be available everywhere, including the matlab module.  

This might be an opportunity to also invent a more pythonic additional long name for them as well, e.g. `zero_pole_gain_to_tf`, but they are so apt and convenient I am not sure that is necessary. 

Feel free to contribute a [pull request](https://github.com/python-control/python-control/wiki/How-to-contribute-with-a-pull-request).

murrayrm, 2022-07-16:
Whoever takes on implementing this should make sure that `zpk2tf` and `zpk2ss` eventually create a `NamedIOSystem`, versus a lower level `TransferFunction` or `StateSpace` object.  That will allow the output system to be compatible with the I/O system functionality (including named input/output signals for connecting things together).  A diagram of the dependencies can be found [here](https://python-control.readthedocs.io/en/latest/classes.html).

ovega14, 2022-07-21:
> 

Should we import `zpk2tf `and `zpk2ss `from `scipy `into the `LTI `module? If so, then these functions will not directly return a `NamedIOSystem`, nor a `TransferFunction `or `StateSpace `object. As a reminder, these functions return numerator and denominator coefficient arrays of a transfer function given zeroes, poles, and gain. Should we then redefine the two functions (`zpk2tf `and `zpk2ss`) to conform to this expected `python-control` behavior?

bnavigator, 2022-07-21:
Yes functions in the `control` namespace should return a python-control conforming system. Hence @murrayrm's request to return `NamedIOSystem`. This requires a wrapper rather than a simple import.

closed 2022-12-17 by murrayrm