python-control/python-control #754: Bug: Incorrect ValueError: incompatible timebases

dc5048, 2022-08-02:
Using control.series to connect several discrete state-space systems, all with dt=0.05, is resulting in "ValueError:Systems have incompatible timebases" from "common_timebase" in namedio.py line 340 in release 0.9.2.  

> File "/usr/local/anaconda3/envs/c/lib/python3.9/site-packages/control/bdalg.py", line 108, in series return reduce(lambda x, y:y*x, sysn, sys1)
> 
>   File "/usr/local/anaconda3/envs/c/lib/python3.9/site-packages/control/bdalg.py", line 108, in <lambda> return reduce(lambda x, y:y*x, sysn, sys1)
> 
>   File "/usr/local/anaconda3/envs/c/lib/python3.9/site-packages/control/iosys.py", line 184, in __mul__ newsys = InterconnectedSystem(
> 
>   File "/usr/local/anaconda3/envs/c/lib/python3.9/site-packages/control/iosys.py", line 895, in __init__ dt = common_timebase(dt, sys.dt)
> 
>   File "/usr/local/anaconda3/envs/c/lib/python3.9/site-packages/control/namedio.py", line 340, in common_timebase raise ValueError("Systems have incompatible timebases")
> 
> ValueError: Systems have incompatible timebases

In my case, the calling function __init__ in iosys.py, line 895 is initializing "dt" as 0 from the global default then passing this as the first arg to common_timebase, while the second arg is the sys.dt from the first system in the list of systems being combined.  This results in the ValueError.  All systems in syslist have dt = 0.05 and pass common_timebase with each other when passed as args manually.

It seems like the intent for the object __init__ functions in this module is probably that dt should be initialized as the dt of the first system in the syslist, but that is not what the code is doing.

As a work-around, running the same script with release 0.9.1 works.

sawyerbfuller, 2022-08-03:
> line 108, in return reduce(lambda x, y:y*x, sysn, sys1)

Thanks for the pointer. Sounds like a bug might have been introduced in a
change to underlying code for how state-space systems are handled. Could
you post code that reproduces the bug?

S

dc5048, 2022-08-03:
The following prints a 4-state state-space object in 0.9.1, but fails with ValueError in 0.9.2:

```
import control

def main():
    ''' recreate discrete state-space bug '''
    A = [[1, 1], [0, 1]]
    B = [[0], [1]]
    C = [[1, 0]]
    D = 0

    sys1 = control.ss(A,B,C,D,0.05)
    sys2 = control.ss(A,B,C,D,0.05)

    sys3 = control.series(sys1, sys2)
    print(sys3)

if __name__ == '__main__':
    main()
```

bnavigator, 2022-08-06:
Fixed by #755. Thanks @murrayrm !

closed 2022-08-06 by bnavigator

Ryton, 2022-11-04:
this still seems broken in 
    __commit__ = 'g3b5f199'
VERSION
    0.9.2
    
As the minimal example throws an error for me.

bnavigator, 2022-11-04:
I cannot find a commit g3b5f199. Where did you install python-control from?

The fix has been merged into the main branch with b9cf384e57a70e2994878c51a0650224afef596a in August. It is not part of the 0.9.2 release as that one was in May. The next release will have the fix.

Ryton, 2022-11-09:
Installed it through anaconda 
'''conda install -c conda-forge control slycot'''

print(control.__version__) gives: 0.9.2
conda list gives: 
control                   0.9.2              pyhd8ed1ab_0    conda-forge

No worries, just mentioning it, bc the issue was closed, i thought it should've been included already in the release.  
But i can pull the main repo cherrypick the fixes.