python-control/python-control #756: CI tests are failing (Slycot issue?)

murrayrm, 2022-08-04:
Conda-based CI tests are failing, as seen in PR #755.  This this is new behavior.  I checked on my local branch and this appears to be happening on the main branch => appears that the issue is outside of this repository.  One possibility is an issue with slycot.

murrayrm, 2022-08-04:
Doing a bit of digging.  It doesn't look like slycot is the issue, since [Python 3.7 without Slycot without cvxopt](https://github.com/python-control/python-control/issues/756#logs) also crashes.  The error messages seem to point to coveralls (?).  I'm try to isolate things [here](https://github.com/murrayrm/python-control/tree/cidebug-04Aug2022).

bnavigator, 2022-08-04:
I don't think that it is coverage.

Looks Qt related which is the default backend for Matplotlib. My first guess is that this is a conda-forge packaging bug on the MPL or Qt side.

More things to try:
- pytest-xvfb
- set `MPLBACKEND`
- try `QT_QPA_PLATFORM=offscreen`

murrayrm, 2022-08-04:
Using `pytest-xvfb` fixes the original crash (which was happening in `config_test.py`, but introduces a new test failure:
```
def test_root_locus_zoom(self):
[102](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:103)
        """Check the zooming functionality of the Root locus plot"""
[103](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:104)
        system = TransferFunction([1000], [1, 25, 100, 0])
[104](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:105)
        plt.figure()
[105](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:106)
        root_locus(system)
[106](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:107)
        fig = plt.gcf()
[107](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:108)
        ax_rlocus = fig.axes[0]
[108](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:109)
    
[109](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:110)
        event = type('test', (object,), {'xdata': 14.7607954359,
[110](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:111)
                                         'ydata': -35.6171379864,
[111](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:112)
                                         'inaxes': ax_rlocus.axes})()
[112](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:113)
        ax_rlocus.set_xlim((-10.813628105112421, 14.760795435937652))
[113](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:114)
        ax_rlocus.set_ylim((-35.61713798641108, 33.879716621220311))
[114](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:115)
>       plt.get_current_fig_manager().toolbar.mode = 'zoom rect'
[115](https://github.com/murrayrm/python-control/runs/7679958935?check_suite_focus=true#step:5:116)
E       AttributeError: 'NoneType' object has no attribute 'mode'
```

There is a note saying that we were using `xvfb-run` to get the right MPL back end, so that is probably related somehow.

murrayrm, 2022-08-04:
One thing that is odd is that the other test suites pass and they are using `xvfb-run`.  In particular, `control-slycot-src.yml` seems fine and it runs the full test suite...

bnavigator, 2022-08-04:
> There is a note saying that we were using `xvfb-run` to get the right MPL back end, so that is probably related somehow.

Yes, it is related. The plain Agg backend does not have the toolbar for the figure manager. We have seen this before: https://github.com/python-control/python-control/pull/504#discussion_r554623327

So this looks more and more like a qt backend issue in the MPL or Qt packages in conda-forge.

Are you able to determine package version differences in the control-slycot-src vs. the python-package-conda runs?

murrayrm, 2022-08-05:
It looks like the error occurs when the installation uses a specialized conda environment (versus just installing things in the conda base environment).  There are quite a few differences in the various packages that are installed in each case.

bnavigator, 2022-08-06:
Now that the conda  error is resolved, there is a new X11 error on the main branch. Something in the GHA runners must have changed between the check run for #755 and after the merge of it.

bnavigator, 2022-08-06:
The failures with the invalid DISPLAY variable are totally random. We might just have been lucky in the builds for #755 and #757.

https://github.com/bnavigator/python-control/commits/update-CI

closed 2022-08-20 by murrayrm

bnavigator, 2022-11-20:
For the record: the random DISPLAY failure is back. (Maybe due to #796)

PR #757 (2022-08-06): Switch CI to mambaforge and conda-forge channel
I think the failures of #756 are due to a mix of conda packages from the main channel and conda-forge packages pulled in with slycot.

- This kind of reverses #736.
- Use mambaforge for speed
- Tidy up the jobnames a bit so that they are more recognizable in the test tabs.
- Use pytest-cov
- Fixes #756