python-control/python-control #760: interconnect fails with new state-space systems

sawyerbfuller, 2022-08-12:
Example code: 
```python
import control as ct
import control.iosys as io
sys1 = ct.ss(1,1,1,1, inputs='a', outputs='b')
sys2 = ct.ss(1,1,1,1, inputs='b', outputs='c')
interconnected = io.interconnect((sys1, sys2), inputs='a', outputs='c')
```

gives the error: 

```python
File ~/Documents/src/python-control-dev/python-control/control/iosys.py:2507, in interconnect(syslist, connections, inplist, outlist, inputs, outputs, states, params, dt, name, check_unused, ignore_inputs, ignore_outputs, **kwargs)
   2505 connections = []
   2506 for input_sys in syslist:
-> 2507     for input_name in input_sys.input_index.keys():
   2508         connect = [input_sys.name + "." + input_name]
   2509         for output_sys in syslist:

AttributeError: 'StateSpace' object has no attribute 'input_index'
```

Maybe an attribute is not correctly set if `ct.ss` is used to instantiate a state-space system? 

If instead the systems are explicitly turned into io systems the following works:

```python
import control as ct
import control.iosys as io
sys1 = io.ss2io(ct.ss(1,1,1,1), inputs='a', outputs='b')
sys2 = io.ss2io(ct.ss(1,1,1,1), inputs='b', outputs='c')
interconnected = io.interconnect((sys1, sys2), inputs='a', outputs='c')
```

murrayrm, 2022-08-12:
Which version of python-control are you using?  I tested on 0.9.2 (latest release) and HEAD of main and it worked OK.

sawyerbfuller, 2022-08-13:
Ok you're right it does work for me on 0.9.2. But not on what I thought is HEAD of main. Maybe I haven't pulled it correctly. Checking.

sawyerbfuller, 2022-08-13:
OK never mind, looks like i am not pulling correctly, apologies!

closed 2022-08-13 by sawyerbfuller