python-control/python-control #761: Passivity test is ill-conditioned

bnavigator, 2022-08-20:
Whack-a-mole! 

Now this cvxopt test is flakytest_`passivity_test.py::ispassive_edge_cases` :

https://github.com/python-control/python-control/runs/7930805491?check_suite_focus=true#step:6:3450

<details>

```
2022-08-20T10:05:12.4606678Z =================================== FAILURES ===================================
2022-08-20T10:05:12.4607612Z _________________ test_ispassive_edge_cases[test_input2-True] __________________
2022-08-20T10:05:12.4607862Z 
2022-08-20T10:05:12.4608255Z test_input = (array([[-3.e+12,  0.e+00],
2022-08-20T10:05:12.4608627Z        [ 0.e+00, -2.e+12]]), array([[0],
2022-08-20T10:05:12.4609085Z        [1]]), array([[-1,  2]]), array([[0.]]))
2022-08-20T10:05:12.4609312Z expected = True
2022-08-20T10:05:12.4609448Z 
2022-08-20T10:05:12.4609567Z     @pytest.mark.parametrize(
2022-08-20T10:05:12.4609948Z         "test_input,expected",
2022-08-20T10:05:12.4610188Z         [((A, B, C, D*0.0), True),
2022-08-20T10:05:12.4610415Z          ((A_d, B, C, D), True),
2022-08-20T10:05:12.4610640Z          ((A*1e12, B, C, D*0), True),
2022-08-20T10:05:12.4610854Z          ((A, B*0, C*0, D), True),
2022-08-20T10:05:12.4611073Z          ((A*0, B, C, D), True),
2022-08-20T10:05:12.4611293Z          ((A*0, B*0, C*0, D*0), True)])
2022-08-20T10:05:12.4611683Z     def test_ispassive_edge_cases(test_input, expected):
2022-08-20T10:05:12.4611958Z         A = test_input[0]
2022-08-20T10:05:12.4612178Z         B = test_input[1]
2022-08-20T10:05:12.4612610Z         C = test_input[2]
2022-08-20T10:05:12.4612970Z         D = test_input[3]
2022-08-20T10:05:12.4613469Z         sys = ss(A, B, C, D)
2022-08-20T10:05:12.4613753Z >       assert(passivity.ispassive(sys) == expected)
2022-08-20T10:05:12.4613952Z 
2022-08-20T10:05:12.4614077Z control/tests/passivity_test.py:115: 
2022-08-20T10:05:12.4614373Z _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
2022-08-20T10:05:12.4614742Z control/passivity.py:292: in ispassive
2022-08-20T10:05:12.4615096Z     return solve_passivity_LMI(sys, rho=ofp_index, nu=ifp_index) is not None
2022-08-20T10:05:12.4615456Z control/passivity.py:177: in solve_passivity_LMI
2022-08-20T10:05:12.4615750Z     sol = cvx.solvers.sdp(c, Gs=Gs, hs=hs)
2022-08-20T10:05:12.4616316Z /usr/share/miniconda3/envs/test-env/lib/python3.10/site-packages/cvxopt/coneprog.py:4126: in sdp
2022-08-20T10:05:12.4616810Z     sol = conelp(c, G, h, dims, A = A, b = b, primalstart = ps, dualstart = ds, kktsolver = kktsolver, options = options)
2022-08-20T10:05:12.4617643Z /usr/share/miniconda3/envs/test-env/lib/python3.10/site-packages/cvxopt/coneprog.py:1395: in conelp
2022-08-20T10:05:12.4618038Z     misc.update_scaling(W, lmbda, ds, dz)
2022-08-20T10:05:12.4618322Z _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
2022-08-20T10:05:12.4618485Z 
2022-08-20T10:05:12.4618810Z W = {'beta': [], 'd': <0x1 matrix, tc='d'>, 'di': <0x1 matrix, tc='d'>, 'dnl': <0x1 matrix, tc='d'>, ...}
2022-08-20T10:05:12.4619335Z lmbda = <6x1 matrix, tc='d'>, s = <13x1 matrix, tc='d'>
2022-08-20T10:05:12.4619636Z z = <13x1 matrix, tc='d'>
2022-08-20T10:05:12.4619773Z 
2022-08-20T10:05:12.4619885Z     def update_scaling(W, lmbda, s, z):
2022-08-20T10:05:12.4620221Z         """
2022-08-20T10:05:12.4620607Z         Updates the Nesterov-Todd scaling matrix W and the scaled variable
2022-08-20T10:05:12.4621042Z         lmbda so that on exit
2022-08-20T10:05:12.4621247Z     
2022-08-20T10:05:12.4621547Z               W * zt = W^{-T} * st = lmbda.
2022-08-20T10:05:12.4621773Z     
2022-08-20T10:05:12.4622145Z         On entry, the nonlinear, 'l' and 'q' components of the arguments s
2022-08-20T10:05:12.4622629Z         and z contain W^{-T}*st and W*zt, i.e, the new iterates in the current
2022-08-20T10:05:12.4622920Z         scaling.
2022-08-20T10:05:12.4623119Z     
2022-08-20T10:05:12.4623489Z         The 's' components contain the factors Ls, Lz in a factorization of
2022-08-20T10:05:12.4624030Z         the new iterates in the current scaling, W^{-T}*st = Ls*Ls',
2022-08-20T10:05:12.4624341Z         W*zt = Lz*Lz'.
2022-08-20T10:05:12.4624645Z         """
2022-08-20T10:05:12.4624832Z     
2022-08-20T10:05:12.4625008Z     
2022-08-20T10:05:12.4625390Z         # Nonlinear and 'l' blocks
2022-08-20T10:05:12.4625606Z         #
2022-08-20T10:05:12.4626008Z         #    d :=  d .* sqrt( s ./ z )
2022-08-20T10:05:12.4626266Z         #    lmbda := lmbda .* sqrt(s) .* sqrt(z)
[... 169 lines cut ...]
2022-08-20T10:05:12.4697652Z             lapack.gesvd(work, lmbda, jobu = 'A', jobvt = 'A', m = m, n = m,
2022-08-20T10:05:12.4698006Z                 ldA = m, U = s, Vt = z, ldU = m, ldVt = m, offsetS = ind,
2022-08-20T10:05:12.4698319Z                 offsetU = ind2, offsetVt = ind2)
2022-08-20T10:05:12.4698543Z     
2022-08-20T10:05:12.4698741Z             # r := r*V
2022-08-20T10:05:12.4699127Z             blas.gemm(r, z, work, transB = 'T', m = m, n = m, k = m, ldB = m,
2022-08-20T10:05:12.4699410Z                 ldC = m, offsetB = ind2)
2022-08-20T10:05:12.4699675Z             blas.copy(work, r, n = m**2)
2022-08-20T10:05:12.4700021Z     
2022-08-20T10:05:12.4700204Z             # rti := rti*U
2022-08-20T10:05:12.4700602Z             blas.gemm(rti, s, work, n = m, m = m, k = m, ldB = m, ldC = m,
2022-08-20T10:05:12.4700867Z                 offsetB = ind2)
2022-08-20T10:05:12.4701098Z             blas.copy(work, rti, n = m**2)
2022-08-20T10:05:12.4701318Z     
2022-08-20T10:05:12.4701645Z             # r := r*lambda^{-1/2}; rti := rti*lambda^{-1/2}
2022-08-20T10:05:12.4702010Z             for i in range(m):
2022-08-20T10:05:12.4702263Z >               a = 1.0 / math.sqrt(lmbda[ind+i])
2022-08-20T10:05:12.4702566Z E               ZeroDivisionError: float division by zero
2022-08-20T10:05:12.4702754Z 
2022-08-20T10:05:12.4703110Z /usr/share/miniconda3/envs/test-env/lib/python3.10/site-packages/cvxopt/misc.py:628: ZeroDivisionError
```

</details>

_Originally posted by @bnavigator in https://github.com/python-control/python-control/issues/759#issuecomment-1221272177_

bnavigator, 2022-08-20:
Attn @Mark-Yeatman

Mark-Yeatman, 2022-08-20:
I ran the passivity tests for this branch 10 times on my local machine (Windows), couldn't reproduce. We could just remove the test case?

bnavigator, 2022-08-20:
The CI failures are on Ubuntu with conda-forge packages.

The parameter set was included for a purpose. Is it meaningful? Necessary to test?

Mark-Yeatman, 2022-08-20:
Yeah, I was trying to set something of an expectation about the relative scaling / condition number of the A and D matrices. I'll give it a look on ubuntu tonight.

Mark-Yeatman, 2022-08-30:
Looks like this hasn't reared its head again (at least according to the "actions" tab in GitHub), and I still can't reproduce locally. Do we want to close this?

murrayrm, 2022-08-30:
Actually, this error continues to be an issue.  I've had to re-run most of the recent CI runs on failed tests and this was the error that showed up.  I'm not sure there is a fix, but we should at least leave this open to keep track of the fact that there is an outstanding issue.

(I have to admit to being confused about why the error shows up randomly; what is the change in underlying state that leads to an error sometimes but not others?)

bnavigator, 2022-08-30:
The only difference I can imagine is that the job lands on a different worker in the GitHub Actions backend.

murrayrm, 2022-11-20:
I've not seen this issue pop up in a while.  We can probably close unless someone has experienced a problem.

bnavigator, 2022-11-20:
The test ist marked XFAIL and still fails occasionally.

From yesterday: https://github.com/python-control/python-control/actions/runs/3502370893/jobs/5866722552#step:6:997

Mark-Yeatman, 2022-12-12:
Sorry to let this sit so long, made a pull request to fix. https://github.com/python-control/python-control/pull/814

closed 2022-12-17 by murrayrm