python-control/python-control #764: control.matlab.lsim does not work for systems of order > 1

jonschlipf, 2022-08-24:
When calling the control.matlab.lsim with a higher-order system (at least 2x2 matrices), there is a prompt:
```
ValueError: Parameter ``T`` must have same elements as the number of columns in input array ``U`` 
```
Inverting U or T does not help. Instead, it prompts a different error with matrix multiplication.
It works properly when calling control.timeresp.forced_response directly, with the "transpose" flag set to false. 
The error is likely due to the transpose flag, and its implementation throughout the declaration of control.timeresp.forced_response

```
import numpy as np
import control
import control.matlab

time = np.linspace(0.0, 511.0e-6, 512)
DAC = np.sin(time)
ADC = np.cos(time)

input_Kalman = np.transpose(np.concatenate(([[DAC]], [[ADC]]), axis=1)[0])
Af = [[0.45768416, -0.42025511], [-0.43354791, 0.51961178]]
Bf = [[2.84368641, 52.05922305], [-1.47286557, -19.94861943]]
Cf = [[1.0, 0.0], [0.0, 1.0]]
Df = [[0.0, 0.0], [0.0, 0.0]]
ss_Kalman = control.ss(Af, Bf, Cf, Df, 1.0e-6)
x_est, _, _ = control.matlab.lsim(ss_Kalman, input_Kalman, time)
```

bnavigator, 2022-08-24:
Confirmed

```
In [5]: input_Kalman.shape
Out[5]: (512, 2)

In [6]: x_est, _, _ = control.matlab.lsim(ss_Kalman, input_Kalman, time)
Traceback (most recent call last):

  Input In [6] in <cell line: 1>
    x_est, _, _ = control.matlab.lsim(ss_Kalman, input_Kalman, time)

  File ~/src/python-control/control/matlab/timeresp.py:297 in lsim
    out = forced_response(sys, T, U, X0, return_x=True, transpose=True)

  File ~/src/python-control/control/timeresp.py:978 in forced_response
    raise ValueError('Parameter ``T`` must have same elements as'

ValueError: Parameter ``T`` must have same elements as the number of columns in input array ``U``

In [7]: x_est, _, _ = control.matlab.lsim(ss_Kalman, input_Kalman.T, time)
Traceback (most recent call last):

  Input In [7] in <cell line: 1>
    x_est, _, _ = control.matlab.lsim(ss_Kalman, input_Kalman.T, time)

  File ~/src/python-control/control/matlab/timeresp.py:297 in lsim
    out = forced_response(sys, T, U, X0, return_x=True, transpose=True)

  File ~/src/python-control/control/timeresp.py:1007 in forced_response
    U = _check_convert_array(U, legal_shapes,

  File ~/src/python-control/control/timeresp.py:805 in _check_convert_array
    raise ValueError(err_msg_start + err_msg)

ValueError: Parameter ``U``: Wrong shape (rows, columns): (512, 2). Expected: (2, 512).
```

bnavigator, 2022-08-25:
@murrayrm fixed it in #765